pothosware/PothosCore #10: bug with decimal separators due to system locale

rizzir, 2015-02-14:
My systems locale is german and so QT uses commas as decimal separators by default in DoubleSpinboxes. This leads of course to some erronous behaviour for example with the Periodogram or QwT Slider.

A quick fix for that would be to set the locale to C/English so that decimal values are always separated by points, no matter what the locale of the system is.

[image]

guruofquality, 2015-02-15:
Thanks for finding this!

So, numbers in Pothos technically have to be "C locale", but the qdoublespinbox is kind of nice that it uses the user's locale setting. So we should probably want to preserve the user's local setting if possible.

So this is the problem code:
https://github.com/pothosware/pothos-gui/blob/master/EditWidgets/DoubleSpinBox.cpp#L23

```
QString value(void) const
{
return QDoubleSpinBox::text();
}
```

So, I could use return QString::number(QDoubleSpinBox::value()) instead.

The only reason that I used "text()" was to preserve the exact number that the user entered (like 1.00 becomes 0.99999999). Its not that important, it just makes the saved value a little "prettier".

Anyway, I think that the following change will work, and still preserve the value as it was presented in the spin box:
```
QString value(void) const
{
    return QString::number(QDoubleSpinBox::value(), 'f', QDoubleSpinBox::decimals());
```

Can you try this change?

guruofquality, 2015-02-15:
* Fix on maint branch https://github.com/pothosware/pothos/commit/5723550b5ca88374c5a441c3e70528acec27c3ba

I forced my locale to german, and this fix seemed to work, let me know if you have additional issues. Thanks!

closed 2015-02-15 by guruofquality

rizzir, 2015-02-15:
Yes that one fixes the spin box but it also means that I end up with different formats for line entries and spin boxes as you can see on the screenshot below. In my opinion that would be a bad idea.

I think a complete localization would be a tricky task.
One problem that directly arises if you want to parse a line from a line edit is that the comma is used as a group separator there. You would need to use something agnostic, like a semicolon. If you wanted to parse 1,2 as an input of a complex number it would have an undefined meaning otherwise because it could be 1 (real), 2 (imaginary) or just 1.2 (real).

[image]

guruofquality, 2015-02-15:
I agree with the consistency issue. The property edit widgets are supposed to be "C formatted" numbers and expressions.

This change sets CLocal on all edit widgets used in the block properties panel: https://github.com/pothosware/pothos/commit/3dba973d4281c57d325bb21e5c0d01d57ca22f16