pothosware/PothosCore #12: Add directory to link.txt for libmawt.so

TierraDelFuego, 2015-03-06:
For posterity if nothing else.
Add directory to link.txt for libmawt.so

Had to add amd64/xawt dir to link.txt so that libmawt.so would get linked to java.so.
In my case pothosgui brought my system to its knees and produced an error but did start.
Even if I added amd64/xawt to LD_LIBRARY_PATH and then started pothos it was still almost killing my system's responsiveness until it started an then pothos was still sluggish.

guruofquality, 2015-03-06:
Thanks for the report. Usually the problem is that the jni libraries arent in the library path, and this results in a error when loading the java support, but nothing catastrohphic. So thats worrysome to hear.

A few questions:
* Do you have a diff for the CMakeLists.txt, im not sure what to change
* Also, what OS/distro are you using

Note to self: the java support is only partially complete (I need to finish the support for writing blocks in java). So, its not particularly useful at the moment; and if its causing problems, it might be worth removing from the main build. After all - it could be built optionally as an out of tree project.

TierraDelFuego, 2015-03-06:
I edited pothos-java/CMakeFiles/JavaSupport.dir/link.txt and added the path to -Wl

This was allowed the java.so to link to mawt.so 
I only made it work no knowledge went into this

but this was easy to figure compared to gnuradio, it's not possible to make gr build into a non-root env and that's what led me to pothos

guruofquality, 2015-03-07:
Cool!

It looks like you arent the only one to have this issue. I found this comment in someone else's cmake jni project: https://github.com/malaterre/GDCM/blob/master/CMake/FindJNI.cmake#L50

The following change is pushed to the master branch, it should look for libmawt and link against it: https://github.com/pothosware/pothos/commit/409126d7fd488a7cafe6761c44e2bdb44529cbbf

guruofquality, 2015-03-22:
Anyway, I hope that this change https://github.com/pothosware/pothos/commit/409126d7fd488a7cafe6761c44e2bdb44529cbbf fixed the issue, so I am closing this bug.

Also, I should add that java support can now be disabled by passing -DENABLE_JAVA=OFF to cmake at project configuration time.

closed 2015-03-22 by guruofquality