pothosware/PothosCore #19: Web JS Editor

sirinath, 2015-08-11:
Using https://github.com/bpmn-io/diagram-js perhaps you can add a web editor.

guruofquality, 2015-08-16:
The topology class has some JSON hooks which is already a helpful start. A simple web server could handle managing a Topology (or Topologies) using the JSON from the client as a reply/request data. One could build quite an interesting and useful web application simply on top of JavaScript and JavaScript libraries.

I don't personally have the bandwidth to make another graphical interface, but for anyone that does -- I will fully support improving and extending the JSON capabilities of the Topology to help make this possible.

Current JSON hooks in Topology:
* Making a topology with JSON, describes affinity zones, blocks, connections: https://pothosware.github.io/pothos/doxygen/0.2.0/classPothos_1_1Topology.html#a9ff7842389431f23505f59f309205857
* Querying performance stats for monitoring purposes: https://pothosware.github.io/pothos/doxygen/0.2.0/classPothos_1_1Topology.html#abe9be5d9e13a3fcc77270eb1b47a3800
* Dumping the topology structure to JSON, possibly for rendering views: https://pothosware.github.io/pothos/doxygen/0.2.0/classPothos_1_1Topology.html#ad63c41eb7d3dfd530a6020776b79df7b

sirinath, 2015-08-16:
Sounds good. Let me explore this further and get back to you.

sirinath, 2015-08-18:
It is possible to have a DSL for configuration in JSON and YAML. Also support expretions within block (https://github.com/ArashPartow/exprtk)

closed 2015-08-18 by sirinath