pothosware/PothosCore #20: Cloud Execution / Untrusted Execution

sirinath, 2015-08-11:
Add ability to do distributed execution on clouds and also handle the security aspects if the code is from untrusted 3rd parties.

guruofquality, 2015-08-18:
I thought that this would come up eventually.  To put it bluntly, I had to start somewhere, so currently interesting stuff like security and automated load balancing (or at least more automated distribution) is not addressed by the current library. The primary use case for the network-aware topologies are working within a local network, although others may be using it for more. Anyway this is something worth exploring. Here is my thought dump and other considerations:

## The proxy server

The proxy server is a tcp service that runs on a remote node, and spawns processes for performing useful work and instances of processing blocks within those processes. The Topology (given a JSON description) connects to these remote servers and spawns the blocks on the remote server in a programmatic way.

* Some more details here: https://github.com/pothosware/pothos-library/wiki/RemoteControlGuide

### Server spawning and port allocation

The proxy server is spawned on a particular port. Users may configure firewall rules, NAT routes, ssh tunnels. The problem currently is that the topology connects to the server and immediately spawns another one which now serves on the next arbitrary available port. The reason for spawning another server is that remote environments/user IP can crash, and we like to keep the master server up.

Because we spawn new servers on unspecified ports, its hard to do things like make firewall rules. The fix is probably to now spawn new servers in the sub-process, but rather, communicate through the master server using some kind of IPC.

### Adding security credentials

The proxy server should be able to operate over SSL, and in addition, use key-based credentials. The user might provide the login credentials like username and  path to  private key with the remote server URL -- in the JSON description.

## Crossing network boundaries

The Topology class is capable of detecting that a connection between two blocks are on separate hosts, and in response, it spawns and connects network source and sink blocks to bridge the flow across the network. This is automatic, but at the same time, its not very smart. This behaviour relies on 1) a local network, 2) spawning a server on the next available port, 3) its not secure.

This issue was in part made to address some of the details about the connection:
https://github.com/pothosware/pothos-library/issues/67

### Automatic inference of network blocks

I would also like to point out that while automatic connections across a network are really helpful, that this problem can be solved manually by simply describing the network source and sink blocks in the JSON description, rather than inferring them. I bring this up, because more complicated network source and sink blocks can be made that operate over other protocols (zmq) or are secure (ssl), or operate across a mesh. These IPs can be developed independently, and once we see a pattern in use case, then the Topology class can be expanded to infer additional types of automatic network blocks.

### Secure network source and sink

All that being said, its probably at least worth adding ssl to the current network source/sink pair so security can be used if needed. Credentials can be passed to the both blocks on instantiation. Security across flows will probably be part of a design choice/optimization as its not always necessary.

## Launching on the cloud 

TODO...still editing

## Validation of plugins

TODO...still editing

guruofquality, 2015-09-11:
A new issue on the core library -- In reference to "Adding security credentials": https://github.com/pothosware/pothos-library/issues/108