pothosware/PothosCore #27: Topology Builder With Language Bindings

sirinath, 2015-08-18:
Support JVM, .Net, builds with also bindings in other languages like Rust, Julia, D Language, Nim Langage, Red Language, Haskell, OCaml

sirinath, 2015-08-18:
Basically what is needed is a Java, C# and C binding as many of the above languages have C bindings for the topology generator.

guruofquality, 2015-08-19:
Are we talking about language bindings for creating the processing blocks themselves? or language bindings for creating an instance of a Topology, blocks, connections, and executing the design?

FWIW, There are python bindings that allow the creation of processing IP, it should also support creating Topology objects (though untested I believe) https://github.com/pothosware/pothos-python/wiki

I did also start work on a Java toolkit, but its not completed: https://github.com/pothosware/pothos-java/wiki

sirinath, 2015-08-19:
You need a JSON, FBP string to create Topology, blocks, connections. Language bindings to create and execute. Also you need processing blocks.