pothosware/PothosCore #32: Crashing PothosUtil (MSVC)

gvanem, 2015-10-08:
I've built Pothos with success in the past using MSVC 2013. But now it crashes on some unknown place in PothosUtil.exe during a MSbuild using MSVC 2015. Last output before crash:
```
CustomBuild:
  Building Custom Rule F:/gv/dx-radio/Pothos/blocks/event/CMakeLists.txt
  CMake does not need to re-run because F:\gv\dx-radio\Pothos\build\blocks\event\CMakeFiles\generate.stamp is up-to-date.
  Generating EventBlocksDocs.cpp
```

More details from WinDbg:
```
Executable search path is: 
ModLoad: 00f40000 00f6a000   F:\gv\dx-radio\Pothos\build\library\apps\Release\PothosUtil.exe
ModLoad: 778c0000 77a2e000   C:\Windows\SYSTEM32\ntdll.dll
ModLoad: 76ab0000 76bf0000   C:\Windows\SYSTEM32\KERNEL32.DLL
ModLoad: 76470000 76547000   C:\Windows\SYSTEM32\KERNELBASE.dll
ModLoad: 5c210000 5c7fa000   F:\gv\dx-radio\Pothos\build\library\lib\Release\Pothos.dll
ModLoad: 64910000 64a44000   f:\ProgramFiler\Pothos\bin\PocoFoundation.dll
ModLoad: 6ca60000 6cab4000   f:\ProgramFiler\Pothos\bin\PocoUtil.dll
ModLoad: 668d0000 66983000   f:\ProgramFiler\Pothos\bin\PocoNet.dll
ModLoad: 70290000 702c5000   f:\ProgramFiler\Pothos\bin\PocoJSON.dll
ModLoad: 6c9f0000 6ca5d000   C:\Windows\SYSTEM32\MSVCP140.dll
ModLoad: 71fe0000 71ff5000   C:\Windows\SYSTEM32\VCRUNTIME140.dll
ModLoad: 71100000 71104000   C:\Windows\SYSTEM32\api-ms-win-crt-runtime-l1-1-0.dll
ModLoad: 71120000 71124000   C:\Windows\SYSTEM32\api-ms-win-crt-stdio-l1-1-0.dll
ModLoad: 71380000 71383000   C:\Windows\SYSTEM32\api-ms-win-crt-filesystem-l1-1-0.dll
ModLoad: 70fe0000 70fe3000   C:\Windows\SYSTEM32\api-ms-win-crt-heap-l1-1-0.dll
ModLoad: 70d90000 70d95000   C:\Windows\SYSTEM32\api-ms-win-crt-math-l1-1-0.dll
ModLoad: 70db0000 70db3000   C:\Windows\SYSTEM32\api-ms-win-crt-locale-l1-1-0.dll
ModLoad: 660e0000 6614a000   F:\gv\dx-radio\Pothos\build\serialization\Release\PothosSerialization.dll
ModLoad: 65ea0000 65f0b000   f:\ProgramFiler\Pothos\bin\PocoXML.dll
ModLoad: 71f80000 71f84000   C:\Windows\SYSTEM32\api-ms-win-crt-convert-l1-1-0.dll
ModLoad: 710f0000 710f4000   C:\Windows\SYSTEM32\api-ms-win-crt-string-l1-1-0.dll
ModLoad: 70fd0000 70fd3000   C:\Windows\SYSTEM32\api-ms-win-crt-utility-l1-1-0.dll
ModLoad: 76720000 7679c000   C:\Windows\SYSTEM32\ADVAPI32.dll
ModLoad: 72ac0000 72ae0000   C:\Windows\SYSTEM32\IPHLPAPI.DLL
ModLoad: 65c20000 65c91000   C:\Windows\SYSTEM32\MSVCP120.dll
ModLoad: 63500000 635ee000   C:\Windows\SYSTEM32\MSVCR120.dll
ModLoad: 76e30000 76e80000   C:\Windows\SYSTEM32\WS2_32.dll
ModLoad: 70b10000 70b15000   C:\Windows\SYSTEM32\api-ms-win-crt-multibyte-l1-1-0.dll
ModLoad: 71030000 71033000   C:\Windows\SYSTEM32\api-ms-win-crt-time-l1-1-0.dll
ModLoad: 70ff0000 70ff3000   C:\Windows\SYSTEM32\api-ms-win-crt-environment-l1-1-0.dll
ModLoad: 763a0000 76463000   C:\Windows\SYSTEM32\msvcrt.dll
ModLoad: 77570000 775b1000   C:\Windows\SYSTEM32\sechost.dll
ModLoad: 76bf0000 76caa000   C:\Windows\SYSTEM32\RPCRT4.dll
ModLoad: 76550000 76557000   C:\Windows\SYSTEM32\NSI.dll
ModLoad: 72ab0000 72ab8000   C:\Windows\SYSTEM32\WINNSI.DLL
ModLoad: 69720000 697fc000   C:\Windows\SYSTEM32\ucrtbase.DLL
ModLoad: 750c0000 750de000   C:\Windows\SYSTEM32\SspiCli.dll
ModLoad: 750b0000 750ba000   C:\Windows\SYSTEM32\CRYPTBASE.dll
ModLoad: 75050000 750a4000   C:\Windows\SYSTEM32\bcryptPrimitives.dll
(84c.49c): Security check failure or stack buffer overrun - code c0000409 (!!! second chance !!!)
eax=00000001 ebx=00000000 ecx=00000005 edx=7f6ab000 esi=00000000 edi=013152a8
eip=6979c3a2 esp=0106f184 ebp=0106f1b4 iopl=0         nv up ei pl nz na po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000202
ucrtbase!_invoke_watson+0x12:
6979c3a2 cd29            int     29h
```

Note the ` Security check failure or stack buffer overrun` in a *Release* build!

I suspect the cause is the previous MSVC 2013 version is interfering with the new build. I.e.MSVC*120.DLL + UCRT dll in the same process. Should the old version be removed first?

guruofquality, 2015-10-08:
Are these Poco dlls built with MSVC 2013? C++ libraries are ABI incompatible between different releases of MSVC.

```
ModLoad: 5c210000 5c7fa000   F:\gv\dx-radio\Pothos\build\library\lib\Release\Pothos.dll
ModLoad: 64910000 64a44000   f:\ProgramFiler\Pothos\bin\PocoFoundation.dll
ModLoad: 6ca60000 6cab4000   f:\ProgramFiler\Pothos\bin\PocoUtil.dll
ModLoad: 668d0000 66983000   f:\ProgramFiler\Pothos\bin\PocoNet.dll
ModLoad: 70290000 702c5000   f:\ProgramFiler\Pothos\bin\PocoJSON.dll
```

How are you installing Poco? As a separate build and install or the git submodule? Just a guess, but its possible that when you built with MSVC 2015, that the Pothos CMake configure, discovered the installed Poco libraries and decided to use them.

Its worth a try to uninstall the old MSVC 2013 files and reconfigure 2015 and see what happens.

gvanem, 2015-10-08:
> ... discovered the installed Poco libraries and decided to use them.

Yes. I had the old `Poco*.dll` in PATH (built with MSVC 2013).

> Its worth a try to uninstall the old MSVC 2013 files and reconfigure 2015 an

I followed the *Build Guide*, added Poco as a `git submodule`. Building all with MSVC 2015 seems okay. 

Isn't there anything in the configure/build process which figures out if the ABI of Poco and Pothos are incompatible? Comparing `_MSC_VER` in both or something?

guruofquality, 2015-10-08:
Unfortunately, there isn't anything in place to detect the ABI incompatibility.
Some techniques that I have seen:

* rename the dll files based on the MSVC version (boost does this)
* A C API call that queries what MSVC version was used to compile the library (this would happen at runtime and its preferable to a segfault)
* Install a text file with Poco that the build system could parse and detect

The first two require changes to Poco. But the last option is something that I could add to the build system.

BTW, I would provide MSVC2015 installers, but I'm still waiting on Qt to release their MSVC2015 compatible installer.

gvanem, 2015-10-08:
> BTW, I would provide MSVC2015 installers

Please do since the Cmake generated `INSTALL.vcxproj` fails to install all stuff under the same directory tree. E.g. some `*.py` files end up in `f:\PothosTestBlocks`!!?

guruofquality, 2015-10-08:
I think the python blocks are failing to detect the python install path, and its getting set to an empty string. The build system should probably 1) disable the component 2) print a helpful message. I didnt know this could happen. I will try to figure this out out...

If you are not using the Python blocks, they can be disabled by passing this to cmake:
```
-DENABLE_PYTHON=OFF
```

guruofquality, 2015-10-08:
@gvanem  When the project is configured, there is a section printed about python dependencies. Im curious what was printed for you. it will help me debug this.

Example:
```
PYTHON_EXECUTABLE: C:/Python34/python.exe
POTHOS_PYTHON_DIR: ${prefix}/Lib/site-packages
PYTHONLIBS_FOUND: TRUE - 3.4.3
PYTHON_INCLUDE_DIRS: C:/Python34/include
PYTHON_LIBRARIES: C:/Python34/libs/python34.lib
```

gvanem, 2015-10-09:
It is:
```
-- Found Qwt: f:/Qt/Qwt/lib/qwt.lib (found version "6.2.0")
-- QWT_VERSION_STRING: 6.2.0
-- QWT_INCLUDE_DIRS: f:/Qt/Qwt/src
-- QWT_LIBRARIES: f:/Qt/Qwt/lib/qwt.lib
-- Looking for include file alloca.h
-- Looking for include file alloca.h - not found
-- PYTHON_EXECUTABLE: f:/MingW32/msys/bin/python << [1]
-- POTHOS_PYTHON_DIR: ${prefix}/
-- Found PythonLibs: F:/ProgramFiler/Python27/libs/python27.lib (found version "2.7.9+")
-- PYTHONLIBS_FOUND: TRUE - 2.7.9+
-- PYTHON_INCLUDE_DIRS: f:/ProgramFiler/Python27/include
-- PYTHON_LIBRARIES: F:/ProgramFiler/Python27/libs/python27.lib
```

[1] that `python` is just a bash link to my real Python27.

gvanem, 2015-10-09:
But just launching PothosGui.exe and exiting again, triggers another crash:
```
ucrtbase!abort+0x4b
ucrtbase!terminate+0x3b
VCRUNTIME140!FindHandler(struct EHExceptionRecord * pExcept = 0x00f9f6fc, struct EHRegistrationNode * pRN = 0x00f9fcf8, struct _CONTEXT * pContext = 0x00000001, void * pDC = 0x00f9f684, struct _s_FuncInfo * pFuncInfo = 0x6aa07fe4, unsigned char recursive = 0x00 '', int CatchDepth = 0n0, struct EHRegistrationNode * pMarkerRN = 0x00000000)+0xd4
VCRUNTIME140!__InternalCxxFrameHandler(struct EHExceptionRecord * pExcept = 0x00f9f6fc, struct EHRegistrationNode * pRN = 0x00f9fcf8, struct _CONTEXT * pContext = 0x00f9f74c, void * pDC = 0x00f9f684, struct _s_FuncInfo * pFuncInfo = 0x6aa07fe4, int CatchDepth = 0n0, struct EHRegistrationNode * pMarkerRN = 0x00000000, unsigned char recursive = 0x00 '')+0xe7
VCRUNTIME140!__CxxFrameHandler2(struct EHExceptionRecord * pExcept = 0x00f9f6fc, struct EHRegistrationNode * pRN = 0x00f9fcf8, void * pContext = 0x00f9f74c, void * pDC = 0x00f9f684)+0x26
ntdll!ExecuteHandler2+0x26
ntdll!ExecuteHandler+0x24
ntdll!KiUserExceptionDispatcher+0xf
KERNELBASE!RaiseException+0x48
VCRUNTIME140!_CxxThrowException(void * pExceptionObject = 0x00f9fc64, struct _s__ThrowInfo * pThrowInfo = 0x704be120)+0x65
PocoFoundation!Poco::ProcessImpl::killImpl+0x85
PocoFoundation!Poco::Process::kill+0xd
Pothos!Pothos::RemoteServer::RemoteServer+0x1a3
Pothos!Pothos::RemoteServer::operator bool+0x47f
PothosGui+0x8867a
KERNEL32!BaseThreadInitThunk+0x24
ntdll!__RtlUserThreadStart+0x2f
ntdll!_RtlUserThreadStart+0x1b
```

The same `Security check failure or stack buffer overrun`. Since there are no `.pdb` files generated, hard to tell where. Creating `.pdb` for release mode is handy too.

BTW. I see Poco listens on TCPv6 port, but my IPv6 adapter (MS's Teredo stuff) is down.

guruofquality, 2015-10-09:
> [1] that python is just a bash link to my real Python27.

So POTHOS_PYTHON_DIR usually evaluates to a relative path Lib/..., but for some reason on this system it becomes an empty string "". What does this command print?

```f:/MingW32/msys/bin/python -c "from distutils.sysconfig import get_python_lib; print(get_python_lib(plat_specific=True, prefix=''))"```

> [2] this should be the same CMAKE_INSTALL_PREFIX I used elsewhere, no? So why isn't it used.

I see.. this is the install command. And it evaluates to a leading slash. Usually install paths are relative paths. The leading "/" because of the empty string is tricking the install command.

```
install(
            FILES ${POTHOS_PYTHON_UTIL_SOURCES}
            DESTINATION ${POTHOS_PYTHON_DIR}/${POTHOS_PYTHON_UTIL_DESTINATION}
        )

```

So I want to add some code to the build system to handle this empty string case. But I am curious why it happens for you.

guruofquality, 2015-10-09:
> But just launching PothosGui.exe and exiting again, triggers another crash:

I think Qt should have the same problem as Poco in the first crash because its also a C++ library. Its going to be an ABI mismatch. I think you need Qt5 for MSVC 2015 as well.

> The same Security check failure or stack buffer overrun. Since there are no .pdb files generated, hard > to tell where. Creating .pdb for release mode is handy too.

In this case, it may help to use "RelWithDebInfo" for CMAKE_BUILD_TYPE, rather than the default "Release"

> BTW. I see Poco listens on TCPv6 port, but my IPv6 adapter (MS's Teredo stuff) is down.

I recently added IPv6 support to Pothos. The server is probably binding to the link-local device as IPv6. But this should not stop IPv4 connections from continuing to work.

gvanem, 2015-10-09:
> What does this command print?

It prints `Lib\site-packages` (same as my normal `python2 ...` would print).

I could probably put `f:/MingW32/msys/bin` *after* `f:\ProgramFiler\Python27` in my `PATH`. But I fear that would break stuff elsewhere. But hey, I'm on Windows. So why doesn't Cmake look for `python.exe`?

> I think you need Qt5 for MSVC 2015 as well.

I've already built Qt5 using MSVC 2015 (from the Qt repo). But as you said, I wish there was a pre-built version for MSVC 2015. Because Cmake fails to find `Qt5Svg` (not sure why it failed to build with MSVC 2015). I do have `Qt5Svg` for MSVC 2013, but going back to that MSVC version is *not* an option for me. I'm pretty up to my neck with too many compilers here already.

guruofquality, 2015-10-09:
As far as finding the python.exe executable goes, it just using CMake's default FindPythonInterp package. I think the exe extension doesnt matter, its probably just finding the first python in the path.

> It prints Lib\site-packages (same as my normal python2 ... would print).

Whats really vexing me is why the command prints "Lib\site-packages", but the code below is coming up with an empty string. Thats the real problem. One thought is that the mingw python is printing to stderr, but the command below is reading stdout..

https://github.com/pothosware/pothos-python/blob/master/cmake/PothosPythonUtil.cmake#L18
```
execute_process(
    COMMAND ${PYTHON_EXECUTABLE} -c
    "from distutils.sysconfig import get_python_lib; print(get_python_lib(plat_specific=True, prefix=''))"
    OUTPUT_STRIP_TRAILING_WHITESPACE
    OUTPUT_VARIABLE POTHOS_PYTHON_DIR
)
file(TO_CMAKE_PATH "${POTHOS_PYTHON_DIR}" POTHOS_PYTHON_DIR)
message(STATUS "POTHOS_PYTHON_DIR: \${prefix}/${POTHOS_PYTHON_DIR}")
```

> I've already built Qt5 using MSVC 2015 (from the Qt repo). But as you said, I wish there was a
> pre-built version for MSVC 2015. Because Cmake fails to find Qt5Svg (not sure why it failed to
> build [...]

If I understand correctly, you tried to build Qt5 from source, and everything build *except* the Qt5Svg library could not build for some reason. I assumed that Qt5 didn't have a MSVC2015 compatible release yet -- hence the lack of installers. But maybe it builds correctly on a branch. IDK But I would like to figure this out so I can build for msvc2015 as well.

gvanem, 2015-10-09:
> One thought is that the mingw python is printing to stderr, but the command below is
> reading stdout..

Maybe. Anyway, I had to hint Cmake to my Python like this:
```
set OPT=-DPOTHOS_PYTHON_DIR:FILEPATH=f:/ProgramFiler  ^
   -DPYTHON_EXECUTABLE:FILEPATH=f:/ProgramFiler/Python27/python.exe ^
...
cmake %OPT%  -G "Visual Studio 14" ..
```
Also with ` -DCMAKE_BUILD_TYPE="RelWithDebInfo"` which gave a more sensible call-stack on  the crash-at-exit which is still there:
```
ucrtbase!abort+0x4b
ucrtbase!terminate+0x3b
VCRUNTIME140!FindHandler(struct EHExceptionRecord * pExcept = 0x0055f04c, struct EHRegistrationNode * pRN = 0x0055f64c, struct _CONTEXT * pContext = 0x00000001, void * pDC = 0x0055efd4, struct _s_FuncInfo * pFuncInfo = 0x6a96de08, unsigned char recursive = 0x00 '', int CatchDepth = 0n0, struct EHRegistrationNode * pMarkerRN = 0x00000000)+0xd4
VCRUNTIME140!__InternalCxxFrameHandler(struct EHExceptionRecord * pExcept = 0x0055f04c, struct EHRegistrationNode * pRN = 0x0055f64c, struct _CONTEXT * pContext = 0x0055f09c, void * pDC = 0x0055efd4, struct _s_FuncInfo * pFuncInfo = 0x6a96de08, int CatchDepth = 0n0, struct EHRegistrationNode * pMarkerRN = 0x00000000, unsigned char recursive = 0x00 '')+0xe7
VCRUNTIME140!__CxxFrameHandler2(struct EHExceptionRecord * pExcept = 0x0055f04c, struct EHRegistrationNode * pRN = 0x0055f64c, void * pContext = 0x0055f09c, void * pDC = 0x0055efd4)+0x26
ntdll!ExecuteHandler2+0x26
ntdll!ExecuteHandler+0x24
ntdll!KiUserExceptionDispatcher+0xf
KERNELBASE!RaiseException+0x48
VCRUNTIME140!_CxxThrowException(void * pExceptionObject = 0x0055f5b8, struct _s__ThrowInfo * pThrowInfo = 0x6a01f0f8)+0x65
PocoFoundation!Poco::ProcessImpl::killImpl(class Poco::ProcessHandleImpl * handle = 0x02a062c8)+0x82
PocoFoundation!Poco::Process::kill(class Poco::ProcessHandle * handle = 0x029ee8e0)+0xd
...
```
I don't know enough of the Pothos design to know why this throw is so fatal.

> If I understand correctly, you tried to build Qt5 from source, and everything build except 
> the Qt5Svg library could not build for some reason

Yes. But just now I got back to my Qt5-repo and did a `git pull & nmake`. Low and behold , the Qt5Svg*.[dll + cmake]` files got produced. So now I was able to build a functioning (?) PothosGui. 
Screenshot below shows it's running with 61 threads!!

[image]

guruofquality, 2015-10-09:
> I don't know enough of the Pothos design to know why this throw is so fatal.

I think I see whats happening. So PothosGui spawns a scratch process. On exit, the stack unwinds, and the handler for this process destructs. It looks like the destructor is throwing when one of the calls sends the kill signal: ```Poco::Process::kill(ph);``` Destructors should never throw, so this needs a try/catch -- though I have never seen it throw before.

> Yes. But just now I got back to my Qt5-repo and did a git pull & nmake. Low and behold , the 
> Qt5Svg*.[dll + cmake]` files got produced. So now I was able to build a functioning (?)

Awesome to hear. I would like to replicate the Qt5 build for the windows installers. Is this just master branch of Qt5?

> Screenshot below shows it's running with 61 threads!!

I think I know whats happening. So first Qt should have a thread pool with some number of pre-allocated threads.

When the topology is active, the background evaluator is periodically re-doing the topology connections. The topology class uses std::future to do this. So while its definitely not using that many threads at once, it may really be using that many threads for this topology update. I think the process thread count just shows the number of threads spawned within a time window, but these are not continual threads performing the blocks processing.

The good news is that the blocks should not be deactivated or stop streaming while this happens. However, the GUI's evaluator should not being doing a full update on the topology -- especially when nothing as changed. This is just laziness on my part and I plan on improving the background evaluator.

gvanem, 2015-10-10:
>  Is this just master branch of Qt5?

Seems so. `v5.6.0-alpha1` according to some files in my `<QT5_ROOT>/.git`.

guruofquality, 2015-10-12:
> PocoFoundation!Poco::Process::kill(class Poco::ProcessHandle * handle = 0x029ee8e0)+0xd

I believe that this crash is now fixed in master: https://github.com/pothosware/pothos/commit/5f9b5e33f764040e2c5ec2efcbaa74b798d55369

> Seems so. v5.6.0-alpha1 according to some files in my <QT5_ROOT>/.git. 

If 5.6.0 alpha supports MSVC 2015, then maybe there will be Qt5 installers soon...

gvanem, 2015-10-12:
> I believe that this crash is now fixed in master: 5f9b5e3

Good. I can verify that *that* crash is gone. But another one has appeared. Tested with `windbg PothosGui.exe`, then just exit when it finished loading (with the previous `.pth`-file). WinDbg reports:
```
HEAP[PothosGui.exe]: HEAP: Free Heap block 00E4B6F8 modified at 00E4B7A0 after it was freed
(151c.4fc): Break instruction exception - code 80000003 (first chance)
```
And the offending call-stack:
```
ntdll!RtlpBreakPointHeap+0x19
ntdll!RtlpFreeHeap+0x4505a
ntdll!RtlFreeHeap+0x1b6
ntdll!RtlDebugFreeHeap+0x212
ntdll!RtlpFreeHeap+0x44b2f
ntdll!RtlFreeHeap+0x1b6
ucrtbase!free+0x1b
PocoFoundation!std::_Deallocate+0x9
PocoFoundation!std::allocator<std::_Tree_node<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,Poco::Logger *>,void *> >::deallocate+0x9
PocoFoundation!std::_Wrap_alloc<std::allocator<std::_Tree_node<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,Poco::Logger *>,void *> > >::deallocate+0x9
PocoFoundation!std::_Tree<std::_Tmap_traits<std::basic_string<char,std::char_traits<char>,std::allocator<char> >,Poco::Logger *,std::less<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::allocator<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,Poco::Logger *> >,0> >::_Erase+0x88
PocoFoundation!std::_Tree<std::_Tmap_traits<std::basic_string<char,std::char_traits<char>,std::allocator<char> >,Poco::Logger *,std::less<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::allocator<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,Poco::Logger *> >,0> >::clear+0xf
PocoFoundation!std::_Tree<std::_Tmap_traits<std::basic_string<char,std::char_traits<char>,std::allocator<char> >,Poco::Logger *,std::less<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::allocator<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,Poco::Logger *> >,0> >::erase+0x43
PocoFoundation!std::_Tree<std::_Tmap_traits<std::basic_string<char,std::char_traits<char>,std::allocator<char> >,Poco::Logger *,std::less<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::allocator<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,Poco::Logger *> >,0> >::_Tidy+0x14
PocoFoundation!std::_Tree<std::_Tmap_traits<std::basic_string<char,std::char_traits<char>,std::allocator<char> >,Poco::Logger *,std::less<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::allocator<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,Poco::Logger *> >,0> >::{dtor}+0x14
PocoFoundation!Poco::Logger::shutdown+0xc5
PocoFoundation!Poco::AutoLoggerShutdown::~AutoLoggerShutdown+0x35
ucrtbase!<lambda_275893d493268fdec8709772e3fcec0e>::operator()+0x8f
ucrtbase!__crt_seh_guarded_call<int>::operator()<<lambda_995298e7d72eb4c2aab26c0585b3abe5>,<lambda_275893d493268fdec8709772e3fcec0e> &,<lambda_293819299cbf9a7022e18b56a874bb5c> >+0x30
ucrtbase!_execute_onexit_table+0x28
```

Does it give you any hints? 

I do have POCO 1.6 (Git) here. But how can I hint your `CMakeLists.txt` to use this instead of the
*normal* `poco154`?

guruofquality, 2015-10-12:
> Good. I can verify that that crash is gone. But another one has appeared. Tested with windbg
> PothosGui.exe, then just exit when it finished loading (with the previous .pth-file). WinDbg reports:

Thanks for the info. When the saved topology is loaded and then closed right away... sounds familiar. I think this might be a bug that I have been chasing for a while.

> I do have POCO 1.6 (Git) here. But how can I hint your CMakeLists.txt to use this instead of the
> "normal" poco154?

Try passing this (it uses the PocoConfig.cmake files installed w/ Poco 1.6)
```
-DPoco_DIR=$installPrefix/lib/cmake/Poco
```

My windows build scripts for reference: https://github.com/pothosware/PothosSDR/blob/master/BuildPothos.cmake#L53

FWIW, either Poco version has the same basic stuff. The only reason in-tree poco submodule is poco154 is because newer poco requires CMake >= 3.0 and I want to support platforms that still ship with the 2.8 series of CMake. Annoyingly, poco only uses cmake 3.0 for one or two lines that could have been if()/endif() for backwards compatibility.

guruofquality, 2015-10-14:
No fixes yet, just an update:

I think I have replicated the same crash on shut-down regarding the logger. Although not consistently, its been a hard one to track down.

And the lock-up that I mentioned happens when a GUI plotter/widget block is being created at the same time the graph editor is being closed. The threads block each other.

guruofquality, 2015-11-22:
Since the original issue is fixed, I'm adding this to the list of evaluator improvements for the GUI and closing the issue. We are most likely seeing a race on shutdown/cleanup: https://github.com/pothosware/pothos-gui/issues/97

closed 2015-11-22 by guruofquality