pothosware/PothosCore #34: Tests failed after building on OSX

Toontje, 2015-10-19:
Failed 5 out of 123 tests
  FAIL: /proxy/python/tests/test_serialization
  FAIL: /proxy/java/tests/test_reflect
  FAIL: /proxy/java/tests/test_primitive
  FAIL: /proxy/java/tests/test_fields
  FAIL: /proxy/java/tests/test_serialization

Any ideas?

audiofilter, 2015-10-19:
I've seen this too (or some of them)

guruofquality, 2015-10-19:
I haven't actually tested the java bindings on OSX, fortunately they aren't really useful for anything yet. But I will try to look into that. However the 1 python failure might actually be the java test of the same name failing, because of something about library symbols:

I left myself this comment:
```
/*!
 * Define the tokens used to declare a static block's function.
 * The address of the function is used to make the static fixture
 * class unique across the complete library + module address space.
 *
 * Clang can only use static, inline functions of the same name will overlap.
 * MSVC can only use inline because static cannot be used as a template argument.
 * And GCC can handle all possibilities: static, inline, and static inline.
 *
 * The specific reason for this implementation and its caveats
 * is becuase clang will cause symbol overlap for class and functions
 * with the same name across modules, even if they are not exported.
 */
#if defined(__clang__)
#define POTHOS_STATIC_FIXTURE_DECL static
#elif defined(_MSC_VER)
#define POTHOS_STATIC_FIXTURE_DECL inline
#elif defined(__GNUC__)
#define POTHOS_STATIC_FIXTURE_DECL static inline
#else
#define POTHOS_STATIC_FIXTURE_DECL
#endif
```

But that may not be any-more because of the new linker flags for making bundles:
```
if(APPLE)
    #fixes issue with duplicate module registry when using application bundle
    set(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS} -flat_namespace")
    set(CMAKE_MODULE_LINKER_FLAGS "${CMAKE_MODULE_LINKER_FLAGS} -flat_namespace")
endif()
```

guruofquality, 2015-10-19:
Also I should ask, what version of java is installed? I think we require 7 and up.

audiofilter, 2015-10-19:
java version "1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-466.1-11M4716)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-466.1, mixed mode)

Can't remember how this was installed...

Toontje, 2015-10-19:
```
Ton@TM-MBP:/Users/Ton>java -version
java version "1.7.0_45"
Java(TM) SE Runtime Environment (build 1.7.0_45-b18)
Java HotSpot(TM) 64-Bit Server VM (build 24.45-b08, mixed mode)
```

Toontje, 2015-10-19:
Also i don't have a PothosGUI utility.

guruofquality, 2015-10-19:
* I believe I replicated the python failure. Basically, the issue that I mentioned about library symbols just needed to be used somewhere else as well. Should be ok on current master.

* java 1.6 is too old. It still compiles because we dont check the version (we should). But there's some introspection classes that it needs from 1.7 and up.

* Get extra information about the error with:
  ```PothosUtil --self-test1 /proxy/java/tests/test_primitive```

* I will try to give jdk latest 1.8 a try tonight

Toontje, 2015-10-19:
Ton@TM-MBP:/Users/Ton>PothosUtil --self-test1
/proxy/java/tests/test_primitive

Testing /proxy/java/tests/test_primitive...

FAIL!

Exception: Pothos::Testing TestJava.cpp:172 -- Unexpected error after last
checkpoint; Error: Exception: Java::FindClass(java/util/Objects)

Toontje, 2015-10-19:
Re-downloaded the source and re-built. I get to this point:
```
Generating moc_MyPlotUtils.cpp
[ 43%] Built target PothosPlotterUtils_automoc
Scanning dependencies of target PothosPlotterUtils
[ 44%] Building CXX object plotters/CMakeFiles/PothosPlotterUtils.dir/PlotterUtils/MyPlotPicker.cpp.o
In file included from /Users/Ton/pothos/plotters/PlotterUtils/MyPlotPicker.cpp:4:
/Users/Ton/pothos/plotters/PlotterUtils/MyPlotPicker.hpp:6:10: fatal error: 'qwt_plot_zoomer.h' file not found
#include <qwt_plot_zoomer.h>
         ^
1 error generated.
make[2]: *** [plotters/CMakeFiles/PothosPlotterUtils.dir/PlotterUtils/MyPlotPicker.cpp.o] Error 1
make[1]: *** [plotters/CMakeFiles/PothosPlotterUtils.dir/all] Error 2
make: *** [all] Error 2
```
Any ideas?

guruofquality, 2015-10-19:
> /Users/Ton/pothos/plotters/PlotterUtils/MyPlotPicker.hpp:6:10: fatal error: 'qwt_plot_zoomer.h' file not found
#include \<qwt_plot_zoomer.h>

Not sure, can you post the configure output from cmake? It looks like it picked up an older version of qwt on the system. But it should be checking the version and using the installed qwt if possible, and otherwise using an internal copy in a git submodule.

> Ton@TM-MBP:/Users/Ton/pothos/build>sudo port install py27-pyqwt
> Error: Unable to execute port: Can't install qt4-mac because conflicting ports are active: qt5-mac

Good news then. theres no need for python for any of the graphics. So py27-pyqwt isnt a requirement.  Qwt is an optional dependency. Not sure about qt4 vs qt5 conflicting. qt5 is definitely a requirement for the graphics stuff. Homebrew let me have both, not sure about mac ports.

> Error: Exception: Java::FindClass(java/util/Objects)

java/util/Objects isnt present in 1.6, but is in newer javas. I suspect a java configuration problem with multiple versions. I have the same issue myself FWIW, still not sure exactly whats wrong. Keep in mind the java bindings are useless right now, so definitely don't stress. But I do appreciate the reports.

audiofilter, 2015-10-20:
You can have both qt5 and qt4 on Mac, but you may end up with conflicts since it will only link one of those and you may get the wrong set of header files without the correct include paths. 
Fortunately it's easy to remove and reinstall them, so I have done that at times to help make sure it's the conflict that is the problem. You may need to run "brew linkapps qt5" too

Toontje, 2015-10-20:
OK, i am now installing qt4-mac from MacPorts and qt5 from the native QT installer from http://www.qt.io/download-open-source/#section-2. Let's see if that works.
Also, i am using java 7, not 6. See above.
Eehhhmmm....
[image]

guruofquality, 2015-10-20:
> QWT_VERSION_STRING: 
-- QWT_INCLUDE_DIRS: /opt/local/include/qwt
-- QWT_LIBRARIES: /opt/local/lib/libqwt.dylib

Heh, so it looks like QWT_VERSION_STRING passes the ```if (QWT_VERSION_STRING LESS "6.1")``` when its empty. Just for reference, what version is your qwt install and is there a macro in /opt/local/include/qwt/qwt_global.h called ```#define QWT_VERSION_STR```?

 As a side note, I know not everyone does homebrew, but there is a homebrew tap: https://github.com/pothosware/homebrew-pothos/wiki And qt4 and qt5 both come prebuilt in "bottles", so the install went fairly quick.

Toontje, 2015-10-20:
I can see that you include qwt 6 in ../pothos/plotters/qwt-6.1.2/src. Why is it not picking up qwt_plot_zoomer.h from there?
I have qwt5 installed.
I had terrible problems in the past running homebrew and macports side by side so i don't want to go that road again.

guruofquality, 2015-10-20:
The build supports using an installed qwt if the version is acceptable. For some reason the version information isn't getting extracted, so the following build system code falls through incorrectly:
```
if (QWT_FOUND)
    message(STATUS "QWT_VERSION_STRING: ${QWT_VERSION_STRING}")
    if (QWT_VERSION_STRING LESS "6.1")
        set(QWT_FOUND FALSE)
        message(STATUS "Qwt older than v6.1 - using built-in Qwt")
    endif()
else (QWT_FOUND)
    message(STATUS "Qwt not found - using built-in Qwt")
endif (QWT_FOUND)
```

I can change the logic to treat an empty version string as not found -- problem solved. I'm just curious why the version extraction fails. Its as if /opt/local/include/qwt/qwt_global.h seems to be missing ```#define QWT_VERSION_STR``` But I think series qwt5 should have this: https://github.com/sikoragmbh/qwt/blob/master/tags/qwt-5.2.1/src/qwt_global.h#L23

```
diff --git a/plotters/CMakeLists.txt b/plotters/CMakeLists.txt                                                                                                                                       
index d7c260a..d8c40a5 100644                                                                                                                                                                        
--- a/plotters/CMakeLists.txt                                                                                                                                                                        
+++ b/plotters/CMakeLists.txt                                                                                                                                                                        
@@ -46,6 +46,10 @@ find_package(Qwt)                                                                                                                                                                 
                                                                                                                                                                                                     
 add_definitions(-DQWT_DLL)                                                                                                                                                                          
                                                                                                                                                                                                     
+if (NOT QWT_VERSION_STRING)                                                                                                                                                                         
+    set(QWT_FOUND FALSE)                                                                                                                                                                            
+endif()                                                                                                                                                                                             
+                                                                                                                                                                                                    
 if (QWT_FOUND)                                                                                                                                                                                      
     message(STATUS "QWT_VERSION_STRING: ${QWT_VERSION_STRING}")                                                                                                                                     
     if (QWT_VERSION_STRING LESS "6.1") 
```

Toontje, 2015-10-20:
By copying everything from ./pothos/plotters/qwt-6.1.2/src into /opt/localinclude/qwt i got passed the qwt_plot_zoomer.h issue.
Now i get this:
```
Scanning dependencies of target PothosPlotterUtils
[ 44%] Building CXX object plotters/CMakeFiles/PothosPlotterUtils.dir/PlotterUtils/MyPlotPicker.cpp.o
[ 44%] Building CXX object plotters/CMakeFiles/PothosPlotterUtils.dir/PlotterUtils/MyPlotUtils.cpp.o
[ 44%] Building CXX object plotters/CMakeFiles/PothosPlotterUtils.dir/PothosPlotterUtils_automoc.cpp.o
make[2]: *** No rule to make target `/opt/local/lib/libqwt.dylib', needed by `plotters/libPothosPlotterUtils.0.dylib'.  Stop.
make[1]: *** [plotters/CMakeFiles/PothosPlotterUtils.dir/all] Error 2
make: *** [all] Error 2
```

guruofquality, 2015-10-20:
Copying over the headers would at best create a mismatched install of qwt headers and library -- and lots of segfaults. I'm not really sure what of qwt was installed in /opt/local, but at this point its probably best to delete everything qwt in /opt/local, reconfigure pothos, and everything should build just fine with the internal qwt copy.

guruofquality, 2015-10-20:
Quick update:
* Moving the java issue to a new bug tracker: https://github.com/pothosware/pothos-java/issues/4 I wouldnt worry about it, but so we dont forget.
* The symbol related bug that caused the python failure when java was installed is fixed: https://github.com/pothosware/pothos/commit/8aaf336b610bf19e31c08838fe6be75266348a9f
* And just in case the FindQwt has this sort of false positive again, I added a work-around to fall back to the internal copy: https://github.com/pothosware/pothos/commit/8b8ffc80b7b68eeb7eb753983ac50ab0b23e5dad

Toontje, 2015-10-20:
I removed all qwt as you suggested and built from scratch. 
Almost there:
```
[ 85%] Building CXX object python/TestBlocks/CMakeFiles/PythonTestBlocks.dir/ForwarderFactory.cpp.o
[ 86%] Building CXX object python/TestBlocks/CMakeFiles/PythonTestBlocks.dir/SimpleSignalEmitterFactory.cpp.o
[ 86%] Building CXX object python/TestBlocks/CMakeFiles/PythonTestBlocks.dir/SimpleSlotAcceptorFactory.cpp.o
[ 86%] Building CXX object python/TestBlocks/CMakeFiles/PythonTestBlocks.dir/PythonTestBlocksDocs.cpp.o
[ 86%] Linking CXX shared module libPythonTestBlocks.so
[ 86%] Built target PythonTestBlocks
[ 86%] Linking CXX shared module libJavaSupport.so
ld: file not found: @rpath/libawt.dylib for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [java/libJavaSupport.so] Error 1
make[1]: *** [java/CMakeFiles/JavaSupport.dir/all] Error 2
```

guruofquality, 2015-10-20:
It does need JDK for the java bindings. What changed since the last time when java stuff built correctly? Anyway, if you want to skip building java, just pass the following to cmake:
```
cmake -DENABLE_JAVA=OFF
```

Toontje, 2015-10-20:
I removed everything qwt related
I activated the qt5-mac port
I replaced JRE 1.7 with JDK 1.8
I pointed CMAKE_PREFIX_PATH to /opt/local/libexec/qt5-mac
Cleaned out the source directory and pulled down the git again

Results:
```
Failed 5 out of 125 tests
  FAIL: /proxy/java/tests/test_containers
  FAIL: /proxy/java/tests/test_reflect
  FAIL: /proxy/java/tests/test_primitive
  FAIL: /proxy/java/tests/test_fields
  FAIL: /proxy/java/tests/test_serialization
```
But that's a later issue.
It still didn't build PothosGui though. Have to do that manually.
But....
[image]

Toontje, 2015-10-20:
Now i only need to learn how to use this stuff. ;-)
[image]

guruofquality, 2015-10-20:
Glad to see it working

> It still didn't build PothosGui though. Have to do that manually.

What did you mean by manually. Setting the CMAKE_PREFIX_PATH for qt?

> Exception: activateStream failed

Looks like an issue with the SDR Play driver: mir_sdr_Init fails. No idea.

Toontje, 2015-10-20:
I mean that i was expecting, when i build fro mthe pothos main branch, that PothosGui would have been built as well. It doesn't, even though the configure step tells me it should. Or at least i cannot find the binary on my system.
I had to clone the PothosGui git and build the GUI from there.

guruofquality, 2015-10-20:
> I mean that i was expecting, when i build fro mthe pothos main branch, that PothosGui would have been built as well. It doesn't, even though the configure step tells me it should. Or at least i cannot find the binary on my system.
I had to clone the PothosGui git and build the GUI from there.

Ok I get it now.. though I have never seen that happen.  If you happen to rebuild and this is a problem again, send a text dump of the cmake output, make output, and install_manifest.txt Should be easy to figure out.

As far as the original unit test failures, I fixed one of them, and I opened up a new tracker for the java thing. So I'm closing out this issue now.

closed 2015-10-20 by guruofquality