pothosware/PothosCore #43: Failure loading built modules

xloem, 2015-11-01:
gdb output:
```
Starting program: /usr/local/bin/PothosUtil --load-module /usr/local/lib/Pothos/modules/plotters/libWaveMonitor.so
[Thread debugging using libthread_db enabled]Using host libthread_db library "/lib64/libthread_db.so.1".
Loading: /usr/local/lib/Pothos/modules/plotters/libWaveMonitor.so
success!
Inconsistency detected by ld.so: dl-close.c: 764: _dl_close: Assertion `map->l_init_called' failed!                                                                                                                                
[Inferior 1 (process 5673) exited with code 0177]
```

This has happened sporadically for me.  Right now, after rebuilding for Debug, it's happening with all the plotters and nothing else.

guruofquality, 2015-11-01:
Might be Qwt not cleaning up correctly (possibly a static initialized object), esp if its only the plotters. I have actually seen the print before on GUI shutdown, but it didnt harm any functionality.

xloem, 2015-11-02:
I'm looking into this for a bit because it's preventing my plotters from loading.  https://sourceware.org/bugzilla/show_bug.cgi?id=11941 might be related.

xloem, 2015-11-02:
I think the right solution to this would be to change how shared libraries are unloaded (not unloading themselves in dynamically loaded static destructors).  I haven't explored the precise conditions which cause this.

guruofquality, 2015-11-02:
Without being too invasive, the following would fix #44 as well and make it possible to unload manually prior to destruction.

1) Plugin will no longer hold a reference to the module that loaded it, rather...
2) When the module is destructed, it removes all related plugins before unload (like #44, but automatic)
3) PluginLoader::loadModules returns a list of the loaded modules, which the caller should hold on to
4) Pothos::init() (which calls loadModules) now stashes the loaded modules
5) I will create a Pothos::deinit() that unloads the modules --- this will be manually called by the GUI or Util app before exiting. Without the deinit, everything falls back to static destruction as it is now.

xloem, 2015-11-02:
So, I'm doing item 1.  Plugins will still need to reference the module file path in order to still have the same interface.  So getActiveModuleLoading() only needs to contain a std::string at https://github.com/pothosware/pothos/blob/master/library/lib/Plugin/Registry.cpp#L131 .

Is there any reason this object is allocated on the heap with SingletonHolder ?  I would like to just use something like
```
    static std::string sh;
    return sh;
``` 
But I don't understand why SingletonHolder is used, to know it should be removed.

xloem, 2015-11-02:
(an alternative solution would be to reference the module with a weak_ptr, so the plugin can use the whole module object, but isn't responsible for its deletion)

guruofquality, 2015-11-03:
> Is there any reason this object is allocated on the heap with SingletonHolder ? I would like to just 

The SingletonHolder is a thread-safe wrapper for aiding "lazy static initialization". This particular variable is actually protected by a mutex, so I probably copied the SingletonHolder from some nearby code in the plugin registry which did actually benefit from it.

xloem, 2015-11-05:
Resolved by deinit_work

closed 2015-11-05 by xloem