pothosware/PothosCore #45: Infinite loop when SoapySDRServer goes down

xloem, 2015-11-02:
If I halt SoapySDRServer while using it, Pothos spews forever lines like:
```2015-11-01 21:28:55 Pothos.Block.callSlot: SDRSource0[setFrequency]: Exception: SoapyRPCPacker::send() FAIL: send() [32: Broken pipe]```

xloem, 2015-11-02:
This was happening only when it had failed to open the device because it was already in use (SoapyRTLSDR).

guruofquality, 2015-11-02:
Its just printing once a second, right? If so, its probably the background evaluator (it has a 1 second timer when there's no other stimulus).

The evaluator is trying to re-apply the same failed setting. I think what should happen is that when a call fails, it should give up on the block and try to re-init the block from scratch. Once that happens, it will try to make the block, fail when it tries to instantiate the SDR device, and then yield a different more sensible error message.

Related issue, this sounds like something to add to it: https://github.com/pothosware/pothos-gui/issues/97

xloem, 2015-11-02:
It printed many, many times per second and caused the pty it was running in to stop responding to input.

xloem, 2015-11-02:
Sorry, after git pulling updates, it does indeed print only once per second.

guruofquality, 2015-11-06:
This is added to the gui issue tracker now: https://github.com/pothosware/pothos-gui/issues/97

closed 2015-11-06 by guruofquality