pothosware/PothosCore #47: Installation confuses existing outdated poco install

xloem, 2015-11-05:
Pothos installed libpoco on my system, but it didn't enable all the components.  Hence other programs on my system began having problems building.

I think one of these two would be ideal:
- if using intree poco, link it statically and don't install it
- or, enable all features when building intree poco

guruofquality, 2015-11-05:
I see what you are saying. Poco does a good job of setting the so versions properly for their different releases. So several runtime libraries of Poco can coexist -- but the development files are going to clash. Here's a few thoughts and other options.

**1) Static link in-tree poco**
We can actually set POCO_STATIC=TRUE to enable building static libraries. The only problem is the poco154 directory is a complete unmodified copy. So it wants to install headers and so files always as the normal make install behaviour.

**2) Enable all features of in-tree poco**
That's not a bad idea. Currently, components of poco that we dont need are disabled -- just to save compile time. They could be turned on conditionally.

**3) Updating portage?**
However, It does beg the question, if the other applications on your system can build against this newer poco (there is a good chance its backwards compatible). What can we do to get gentoo to update to a recent poco release?

**4) Alternative poco install dir**
Another option could involve installing a recent poco release in a different directory and pointing pothos to it. That way it wont interfere with the default system poco development files.

We can definitely add 2) as a build option. 4) would be a very clean work-around to replace the entire intree copy and all of its associated problems. And I hope 3) is a possibility down the road.

xloem, 2015-11-05:
1) I guess I was assuming that it's possible to come up with some cmake magic to exclude the poco154 subdir when the install target is made.
3) portage update is waiting on an asm issue with 7z which occurs on x86 only: https://bugs.gentoo.org/show_bug.cgi?id=550976#c8 My system is not x86 so I didn't rush into testing it

guruofquality, 2015-11-08:
For 1) I don't know how, but if its possible. But thats something I would like to add. Also, as a alternative, although tedious, someone could make install -C for all subdirectories except poco.

For 2) I added the following: https://github.com/pothosware/pothos/commit/5291e6e46401e8022231797045b1765fa677dca4 Pass -DPOCO_MIN_BUILD when configuring, poco will enable whatever if can when its various dependencies are on the system.

xloem, 2015-11-09:
1) https://cmake.org/cmake/help/v3.4/command/add_subdirectory.html describes `EXCLUDE_FROM_ALL` which only builds the contents of a subdirectory if it is depended on by something else.  It's ambiguous about whether the subdirectory is installed in that case.

xloem, 2015-11-09:
1) https://cmake.org/cmake/help/v3.4/command/add_custom_command.html Another option might be to exclude poco from the build system, but build it with a custom command.

guruofquality, 2015-11-19:
> 1) https://cmake.org/cmake/help/v3.4/command/add_subdirectory.html describes EXCLUDE_FROM_ALL 

Combining EXCLUDE_FROM_ALL with POCO_STATIC was the right idea. Unfortunately it looks like some parts of Poco cant handle being static, such as the logging which relies on internal data structures in the shared poco foundation library.

guruofquality, 2016-03-08:
So we added some build hacks, but ultimately, I recommended getting Poco from the OS's package manager. The included submodules are supposed to be helpful for systems that don't have all of the packages.

Definitely gentoo should get an up to date poco. Poco could also be manually compiled and installed to a non-system directory, point pothos to this version of Poco with -DPoco_DIR, and make sure the so files for this poco are in the LD_LIBRARY_PATH at runtime. Not sure what else I can do. Closing, thanks.

closed 2016-03-08 by guruofquality