pothosware/PothosCore #51: PothosGui: terminate called after throwing an instance of 'Poco::Net::DNSException'   what():  DNS error

TheOncomingStorm, 2015-12-14:
I get this every time I run it. I've built Poco through the subtree as well as downloading the 1.61 tar.gz as well as getting the freshest copy available today from their git and still I get the same error. 

After some research I found this http://pocoproject.org/forum/viewtopic.php?f=12&t=1308&p=2793#p2793 which seems to suggest that the problem might be with Pothos, though it's a different error. 

Another thread suggested it was a problem with the version of gcc I was using but I got the same results regardless of which version I switched to. 

Any help appreciated.

guruofquality, 2015-12-14:
Well, its probably not a problem with poco, I bet there is just a url that doesnt resolve on your system for some reason or another (like localhost or ::1). Also, the exception isnt caught and printed/logged properly. So its probably throwing in an unexpected way.

What OS are you running BTW?

A backtrace
====

My first suggestion is to get a backtrace. This will let me know where its throwing, and what the input was that caused it. I guess launching the gui is causing the exception? try the following

```
gdb PothosGui
# type run + enter at the prompt
>>>run
# when it crashes, get the backtrace
>>>bt
# share the output
```

Networking
====
Aside from that, is there anything interesting network-wise on your system? firewall configured? vpn?

Unit tests
====
It also might be good to know if any of the unit tests crash in a similar way:

```
PothosUtil --self-tests
```

TheOncomingStorm, 2015-12-15:
Well I don't know what happened but it just started working for no reason. I was checking to see if the binary was causing a PAX violation(I'm running a hardened kernel on Gentoo) and it just worked. 

I had checked for localhost and ::1 and they were in my /etc/hosts and then ran the --self-tests and it had said that all 119 tests had passed(though it complained about some plugins not loading safely, I'll check that later). Then I just tried it to see if it was causing that PAX violation and it opened the gui without any idea why. I had re-compiled with --debug but had already ran it once before with no success. Well no idea what fixed it but I'll go ahead and close this. Thanks for the help.

closed 2015-12-15 by TheOncomingStorm

guruofquality, 2015-12-15:
Ok, well good to know. Please let me know if it happens again. Thanks.

> though it complained about some plugins not loading safely, I'll check that later

There was some reorganization for the next release. If you clean lib/Pothos/modules and reinstall it will go away.