pothosware/PothosCore #53: MSVC build broken

gvanem, 2016-01-10:
I used to be able to build Pothos just fine using MSVC 2015 approx 6 months ago. But now there are all kinds of problems. E.g. in `build\gui\CMakeFiles\PothosGui.dir\build.make`, there is:
```
gui\CMakeFiles\PothosGui.dir\PothosGui.rc.res: gui\PothosGui.rc
  ...
     F:\ProgramFiler-x86\MS-SDK-8.1\bin\x86\rc.exe  $(RC_FLAGS) -O coff $(RC_DEFINES) -
  ...
```

`coff` for MSVC??!!

Besides there is tons of CL warnings, missing `-D_USE_MATH_DEFINES`, need to add `<Pothos-rot>\build\poco\lib` to `%LIB%` and lots of:
```
HostInfo.cpp
Unknown compiler version - please run the configure tests and report the results
```

PS. I used:
```
-DCMAKE_BUILD_TYPE="Release" -G "NMake Makefiles"` 
```
(+ a lot more) options in Cmake. And *"RelWithDebInfo"*  no longer seems to work; The `*.make` files are trying to mix *Release* and *Debug*.

guruofquality, 2016-01-10:
There was repository reorganization and I updated the poco submodule to git 1.6.2. Its possible that the poco update is the cause of some problems.

Although I just got a pull request for building PothosSDR environment with MSVC2015, and there weren’t any patches for pothos or poco: https://github.com/pothosware/PothosSDR/pull/12

I will try the entire PothosSDR under MSVC2015 and see what happens.

> coff for MSVC??!!

No idea what this means. I am happy to take suggestions. Its just some code that I found on the internet to build an icon into the PothosGUI executable. I have never had an issue with it in the past. And I haven't changed it since it was added, AFAIK.

```
########################################################################
# Resource file - adds an icon to Pothos GUI executable
########################################################################
if (WIN32)
    set(ICON_SOURCE ${CMAKE_CURRENT_SOURCE_DIR}/icons/PothosGui.ico)
    set(RES_FILES "${CMAKE_CURRENT_BINARY_DIR}/PothosGui.rc")
    file(WRITE "${RES_FILES}" "id ICON \"${ICON_SOURCE}\"")
    set(CMAKE_RC_COMPILER_INIT windres)
    enable_language(RC)
    set(CMAKE_RC_COMPILE_OBJECT
        "<CMAKE_RC_COMPILER> <FLAGS> -O coff <DEFINES> -i <SOURCE> -o <OBJECT>")
    list(APPEND SOURCES ${RES_FILES})
endif(WIN32)
```

> -DCMAKE_BUILD_TYPE="Release" -G "NMake Makefiles"` 

I haven't used NMake. Perhaps some of these problems like the compiler identification message and the RelWithDebInfo not working be caused by the differences with the NMake build?

gvanem, 2016-01-11:
>> coff for MSVC??!!

> No idea what this means. I am happy to take suggestions. 

Your rule should consider `MSVC`:
```
if (MSVC)
  set(CMAKE_RC_COMPILE_OBJECT
          "<CMAKE_RC_COMPILER> <DEFINES> <SOURCE> -fo <OBJECT>")
else (WIN32)
 as above
```
(untested).

guruofquality, 2016-01-12:
Just to confirm, everything should build with cmake and MSVC **project files** and visual studio 2015. And I just released new installers for version 2015.

As far as this issue is concerned:
* The NMake build used to work 6 months ago, and now it doesn't. Is that correct?
* Is it important to support NMake now that there are 2015 installers?
* Whats the main use case for nmake here (just curious): for building the pothos runtime and libraries? or was or custom projects with blocks etc?

> Your rule should consider MSVC:

OK, I can see that there are other compilers for windows that will not have the rc compiler (like cygwin). So thats a good change.

gvanem, 2016-01-12:
> The NMake build used to work 6 months ago, and now it doesn't. Is that correct?

I've never used the Cmake *NMake Makefiles* generator in Pothos until now. I used to use the *Visual Studio 14*  generator with success in the past. Why this is not working, I've no idea. I've updated Cmake to v3.4.0 in the mean-time though (if that matters).

> Is it important to support NMake now that there are 2015 installers?

Besides **I don't trust software I cannot build myself**? :smiley: I suck at Cmake, but I assume this build-system and your `CMakeLists.txt` are *generator agnostics*. So the binary result should pretty much be the same regardless of how one builds this. Or?

I didn't know there is a 2015 installer. I'll check it out. Built using a recent Qt5 I assume?

guruofquality, 2016-01-14:
I have never actually gotten the NMake generators to work myself. I only every build and test Unix makefiles and visual studio project files. My assumption is that nmake generated files would just invoke the cl.exe compiler with the same basic flags and files that a visual studio project would. But I don't know if cmake gives equal treatment for nmake support.

> I didn't know there is a 2015 installer. I'll check it out. Built using a recent Qt5 I assume?

Just released, and its using a beta qt 5.6: https://groups.google.com/forum/#!topic/pothos-users/1HOKTbOPHpo

If it makes life any easier, you can replicate the build for everything in the installer. PothosSDR is just a cmake project that clones from git and builds other cmake projects with visual studio. And to simplify the build, just comment some things out that are not needed, like hardware support, various toolkits, their dependencies. I hope it helps.

* https://github.com/pothosware/PothosSDR/blob/master/README.build
* https://github.com/pothosware/PothosSDR/blob/master/README.md#prebuilt-binaries-manifest

guruofquality, 2016-02-02:
I guess everything is ok as long as vs project generators are used. Closing.

closed 2016-02-02 by guruofquality