pothosware/PothosCore #54: Bundled GNURadio + GrOsmoSDR (PortAudio version)

veso266, 2016-01-12:
Hi after I've updatet to newest version I can't open gnuradio anymore and hackrf_info shows nothing at all
[image]

it is probably config related I don't know where to put config file now (is it still in: C:\Users\Mitja\AppData\Roaming\.gnuradio\config.conf?)

Thanks for Anwsering and Best Regards

guruofquality, 2016-01-12:
The libusb version and hackrf utils should not have changed at all. I haven't tested all of the installers specifically, but the "PothosSDR-2016.01.11-vc14-x64.exe" found my hackrf just fine. Which installer did you use? Perhaps a usb driver/zadig configuration problem?

As far as the import error. Something isnt set, like PATH or PYTHONPATH. Not sure, the GNURadioHelper.py script can help to diagnose this: https://github.com/pothosware/PothosSDR/wiki/GNURadio#install-dependencies-automatic

One more thing that I screwed up (minor). The installer used to set GRC_BLOCKS_PATH environment variable. I thought that the prefs fixes for gnuradio would make this unnecessary, but it turns out that GRC does not use the prefs. :-(

* If you install to c:\PothosSDR, GRC will work no problem (good)
* Otherwise set GRC_BLOCKS_PATH to c:\install_prefix\share\gnuradio\grc\blocks
* Or use the prefs:

```
[grc]
global_blocks_path = c:/install_prefix/share/gnuradio/grc/blocks
```

veso266, 2016-01-13:
Hi your script found the problem but if I add C:\Program Files (x86)\gnuradio\lib\site-packages to PYTHONPATH (I already did that) then the error is the same and script tells me to do it again
Here is the output from the script: python GNURadioHelper.py > error.txt

========================================
== Runtime and import checks
========================================
 * Check Python version is 2.7...    PASS
 * Check locate GTK+ runtime...      PASS
 * Check import gtk module...        PASS
 * Check locate GNURadio runtime...  PASS
 * Check import GNURadio module...   FAIL
 * Check import numpy module...      PASS
 * Check import lxml module...       PASS
 * Check import Cheetah module...    PASS
 * Check import wx module...         PASS
 * Check import OpenGL module...     PASS

========================================
== Checks passed summary
========================================
PYVERSION:	2.7.10 (default, May 23 2015, 09:44:00) [MSC v.1500 64 bit (AMD64)]
GTK_RUNTIME:	C:\Program Files\GTK2-Runtime Win64\lib\libgtk-win32-2.0-0.dll
IMPORT_GTK:	C:\Python27\lib\site-packages\gtk-2.0\gtk\__init__.pyc
GR_RUNTIME:	C:\Program Files (x86)\gnuradio\bin\gnuradio-runtime.dll
IMPORT_NUMPY:	C:\Python27\lib\site-packages\numpy\__init__.pyc
IMPORT_LXML:	C:\Python27\lib\site-packages\lxml\__init__.pyc
IMPORT_CHEETAH:	C:\Python27\lib\site-packages\Cheetah\__init__.pyc
IMPORT_WX:	C:\Python27\lib\site-packages\wx\__init__.pyc
IMPORT_OPENGL:	C:\Python27\lib\site-packages\OpenGL\__init__.pyc

========================================
== Checks failed summary
========================================
IMPORT_GR:	DLL load failed: Navedenega modula ni bilo mogoče najti.

========================================
== Fixing problems
========================================
Handling issues for IMPORT_GR...
Error: GNURadio modules missing from PYTHONPATH
  Add C:\Program Files (x86)\gnuradio\lib\site-packages to the PYTHONPATH

PS: I've used this installer: PothosSDR-2016.01.11-vc14-x64.exe
PPS: if I compare previus and this installer with TotalCommander you can see that there are some files missing in new installer:
[image]
look at green files (I suspect theese files are the reason why hackrf_info failed to run

guruofquality, 2016-01-13:
> Handling issues for IMPORT_GR...
Error: GNURadio modules missing from PYTHONPATH
Add C:\Program Files (x86)\gnuradio\lib\site-packages to the PYTHONPATH

My recommendation is to see what python thinks the path is. I'm not sure why python is confised. Maybe a space or quote issue? Run the following command, below is an example. You should see ```C:\Program Files (x86)\gnuradio\lib\site-packages``` appear in the print below if the path is correct.

```

c:\>c:\Python27\python.exe -c "import sys; print '\n'.join(sys.path)"

c:\PothosSDR\lib\site-packages
C:\Windows\SYSTEM32\python27.zip
c:\Python27\DLLs
c:\Python27\lib
c:\Python27\lib\plat-win
c:\Python27\lib\lib-tk
c:\Python27
c:\Python27\lib\site-packages
c:\Python27\lib\site-packages\gtk-2.0
c:\Python27\lib\site-packages\wx-3.0-msw
```

Also FYI. Typically 32-bit programs go in ```c:\Program Files (x86)```. However this suite is actually a 64-bit program and so ```c:\Program Files``` is recommended. Although it still should work, I am curious if the 64-bit python was ignoring paths found in ```c:\Program Files (x86)```. Just a guess, but if you continue to have trouble with PYTHONPATH, then please give ```c:\Program Files``` a try.

> PS: I've used this installer: PothosSDR-2016.01.11-vc14-x64.exe

I tested this exact installer on two machines now. Both work. I have a few ideas:

* Please double check windows device manager, make sure that the hack rf is present and that there is no warning/error symbol on the hackrf.

* Does switching back to older PothosSDR fix the problem? I just want to confirm its the software and not a windows/libusb driver issue.

* When I tested on a virgin machine, I got the error "msvcr100.dll" is missing. I believe that the pthread or libusb precompiled binaries were built with msvc 2010. So it looks like visual studio runtime 2010 is actually an unstated requirement (its small) http://www.microsoft.com/en-us/download/details.aspx?id=13523 Typically visual studio runtimes are already installed by another program. If not, please try installing "Microsoft Visual C++ 2010 SP1 Redistributable Package (x64) " Perhaps this is why hackrf_info simply crashed for you without the warning that I got. Just a guess.

* We don't know if hackrf_info is crashing or simply not finding any devices. It may be worthwhile to try dependency walker on hackrf_info.exe. This would confirm the msvcr100.dll suspicion as well: http://www.dependencywalker.com/

> PPS: if I compare previus and this installer with TotalCommander you can see that there are some files missing in new installer:

No worries. That looks expected.

* boost*.dll files should have "1_60" for version 1.60 now. So the file name is different
* msvc.dll files should have "140" for version vc14 (aka msvc 2015)
* And good news, the arispy patch was sent to upstream airspy.git :-)

guruofquality, 2016-01-14:
* Little update: The helper script (https://github.com/pothosware/PothosSDR/blob/master/Scripts/GNURadioHelper.py) now sets the PYTHONPATH and GRC_BLOCKS_PATH to the current user's environment. If there was just an accidental typo, the script should set the correct path.

* I see now that hackrf_info prints a message when no devices are found.. so It looks like hackrf_info may really be crashing for you. I hope this is a case of missing dll like msvcr100.dll

pavel-demin, 2016-01-14:
I've just tried the latest version of the helper script and found the following problems:
 - If I set `PYTHONPATH` to `C:\Program Files\PothosSDR\lib\site-packages` before running the helper script, then it converts it to `C:\Program;Files\PothosSDR\lib\site-packages` (with a semicolon between Program and Files). I'd say that the commands at lines 176 and 181 are doing this conversion:
https://github.com/pothosware/PothosSDR/blob/master/Scripts/GNURadioHelper.py#L176
https://github.com/pothosware/PothosSDR/blob/master/Scripts/GNURadioHelper.py#L181

 - `check_import_gr` does not work because gnuradio requires numpy and the script just stops at this check and never installs numpy. As a workaround, I've removed `("IMPORT_GR",      'import GNURadio module',  check_import_gr, handle_import_gr)` from my copy of the script. Maybe putting it after `("IMPORT_NUMPY",   'import numpy module',     check_import_numpy, handle_import_numpy)` would solve this problem.

 - `pip install cheetah` does not work on my system. It crashes at `building 'Cheetah._namemapper' extension` because it can't find a C compiler. Installing a binary package (from http://www.lfd.uci.edu/~gohlke/pythonlibs/#cheetah) solves this problem.

guruofquality, 2016-01-14:
> then it converts it to C:\Program;Files\PothosSDR\lib\site-packages

Doh. It should have been split(';'). Thanks for finding that

> check_import_gr does not work because gnuradio requires numpy

I didn't realize that numpy was required for the gr module itself. This never gave me an issue when I tried it on a vanilla system. Strange.. anyway. I reordered the checks for this dependency.

> pip install cheetah does not work on my system.

Thats new... It used to never compile cheetah. Always using the slow python-based namemapper (something like that). Anyway, easy fix. It now should download the .whl mentioned instead.

pavel-demin, 2016-01-15:
Now, GNURadioHelper.py works flawlessly on my system. PortAudio and GrOsmoSDR also work great. Thanks a lot for bringing all these great software tools to Windows!

pavel-demin, 2016-01-15:
>> pip install cheetah does not work on my system.

> Thats new... It used to never compile cheetah. Always using the slow python-based namemapper (something like that). Anyway, easy fix. It now should download the .whl mentioned instead.

I've just tested it with a few versions of Python 2.7. `pip install cheetah` works with Python 2.7.10 (32-bit) and Python 2.7.11 (32-bit) but it doesn't work with the 64-bit versions. So, something is not working properly with the 64-bit versions.

Looks like msvc9compiler.py raises an exception that cheetah installer doesn't catch:
https://github.com/rtyler/cheetah/blob/master/SetupTools.py#L31 
```
    building 'Cheetah._namemapper' extension
    Traceback (most recent call last):
      ...
      File "D:\Test27\lib\distutils\msvc9compiler.py", line 299, in query_vcvarsall
        raise ValueError(str(list(result.keys())))
    ValueError: [u'path']
```
After looking at the code in msvc9compiler.py, I'd say that the problem is that I have Visual Studio C++ 2008 Express Edition with only a 32-bit compiler. In this case, msvc9compiler.py raises  ValueError that cheetah/SetupTools.py does not expect.

guruofquality, 2016-01-15:
Thanks for looking into it. In the very recent past I have installed cheetah on a 64-bit python without any need for compilation tools. So maybe this is a new change for the pip package. In any case, I think that the binary whl file is a better alternative.

veso266, 2016-01-15:
Hi GNURadio works now I just use vc12 installer and uninstall everything then install it and run GNURadioHelper.py to configure everything and then it worked the onlything that doesn't work is PortAudio my config.conf is still in C:\Users\Mitja\AppData\Roaming\.gnuradio and it has that in: 

## This file contains system wide configuration data for GNU Radio.
## You may override any setting on a per-user basis by editing
## ~/.gnuradio/config.conf

## specify which audio module to load, or use "auto" to have the system
## select one.  Valid choices depend on your OS and which modules
## you've installed, but typically include:
## auto, alsa, oss, portaudio, jack, osx, windows

[audio]

audio_module = portaudio

but portaudio still doesn't work GNURadio still defults to windows architure
[image]

PS: My default recording source is Line In (from my Sony XDR-F1HD)

Thanks for Anwsering and Best Regards

pavel-demin, 2016-01-15:
I have config.conf with almost  the same content (just without any comments) and at the same location:

```
AppData\Roaming\.gnuradio\config.conf
```

```
[audio]

audio_module = portaudio

```
My PothosSDR version is PothosSDR-2016.01.11-vc12-x64.exe.

Both 'Audio Source' and 'Audio Sink' work:
[image]

veso266, 2016-01-15:
hmm thats strange so why does it defult to windows if I have everything in the config set corecly?
(I've removed all the coments from config and (just incase (I tried without it) I changed: C:\Program Files (x86)\gnuradio\etc\gnuradio\conf.d\gr-audio.conf) to the same as config.conf

Thanks for Anwsering and Best Regards

pavel-demin, 2016-01-15:
> `C:\Program Files (x86)\gnuradio\etc\gnuradio\conf.d\gr-audio.conf`

On my system this file is in `C:\Program Files\PothosSDR\etc\gnuradio\conf.d\gr-audio.conf`

I've just noticed the following line in your first comment:
> `GR_RUNTIME: C:\Program Files (x86)\gnuradio\bin\gnuradio-runtime.dll`

Here is how it looks like on my system:
`GR_RUNTIME: C:\Program Files\PothosSDR\bin\gnuradio-runtime.dll`

Is it possible that there are two GNU Radio versions installed on your system?

veso266, 2016-01-15:
No as I am aware there is just one version of GNU Radio installed (I just renamed the folder to gnuradio instead of PothosSDR)

guruofquality, 2016-01-15:
> I just renamed the folder to gnuradio instead of PothosSDR

Lol why? I bet this rename is the cause of the problem.

The installer sets several environment variables so various libraries in PothosSDR/ can find their files. In the case the gnuradio, if the install prefix directory is not found (this is set by environment variable on install) gnuradio will not load any configuration files, not even the ones in %appdata%. If you just rename the directory, the environment variable set by the installer will tell it to look in the old location.

veso266, 2016-01-15:
I've renamed the folder from gnuradio to PothosSDR changed all PATHs to reflect PothosSDR instead of gnuradio and moved everything from Program Files (x86) to Program Files but it still want to use arch: windows not portaudio

Thanks for Help

guruofquality, 2016-01-15:
I know that it can be made to work. But given all of the renaming and confusion, I highly recommend uninstalling/deleting everything and starting from a clean install. There are too many changes going on. Things get complicated. Its best to start from a clean install before making big changes.

* run the PothosSDR uninstaller.exe
* delete any related files that remain
* Use rapid environment editor and delete any environment variable with the words:
  * POTHOS
  * UHD
  * GRC
  * GNURADIO
  * PYTHONPATH
* Restart PC
* Reinstall PothosSDR
* Run GNURadioHelper.py

Now hopefully it works. If not, below I have some debugging tips:

Debugging tips
====

Double check that gnuradio-config-info has the correct install prefix. The resulting directory should match the %POTHOS_ROOT% environment variable and should exist. If not, the config will not be loaded.
```
C:\>gnuradio-config-info --prefix
C:\Program Files\PothosSDR
```

Gnuradio does not have an easy way to print the config files that it searches. But there is a programmatic way to print the current configuration. This may be easier than running grc to test working:
```
C:\>c:\Python27\python.exe -c "from gnuradio import gr; print gr.prefs().to_string()"
[audio]
audio_module = portaudio

[audio_portaudio]
verbose = false

etc...
```

veso266, 2016-01-16:
Thanks it worked (I uninstall everything then reinstall it and I discovered that config wasn't loaded)

Thanks for Helping

guruofquality, 2016-01-16:
Glad to hear its working! Thanks @veso266 @pavel-demin 

> PS: to use specific device just put its name in Device Arguments: (my was: Line 1 (Virtual Audio Cable))

If you learn any helpful tips, like audio configuration, etc. Can you add them to this section of the wiki? https://github.com/pothosware/PothosSDR/wiki/GNURadio#application-notes (I think its public write access)

> PPS: Is posible to install some OTA Projects like gr-rds: https://github.com/bastibl/gr-rds

Sure, its possible. Keep in mind that this is uncharted territory for many of these out of tree projects. There is usually a lot of patching to get things compiling at the least. If you have a wishlist suggestion, please:

  * Make an issue on the tracker: https://github.com/pothosware/PothosSDR/issues
  * Give me a link to the best github repo to use (most up to date/best maintained/most official)
  * Maybe give a priority like low or high.. so I know which projects to try and build first

closed 2016-01-16 by guruofquality