pothosware/PothosCore #70: module load can fail in regular load -- but not safe load

guruofquality, 2016-02-27:
_From @guruofquality on July 25, 2014 21:58_

If there is for example, a plugin registry overlap, the safe load is OK, its only loading one module, never sees the overlap...

However, the regular load will throw in a static init call, and bad things happen...

See if we can at least manage the error better...

_Copied from original issue: pothosware/pothos-library#54_

lrmodesgh, 2018-01-15:
i do get the similar error after installing pothos from source. i am pasting the console o/p here: 

> 2018-01-15 23:25:04 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libLogicAnalyzer.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libSpectrogram.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libConstellation.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libWaveMonitor.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libPeriodogram.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libQwtWidgets.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libLogicAnalyzer.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libSpectrogram.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libConstellation.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libWaveMonitor.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libPeriodogram.so): failed safe load
> 
> 2018-01-15 23:25:05 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libQwtWidgets.so): failed safe load
> 
> 2018-01-15 23:25:05 PothosFlow.MainWindow: Welcome to Pothos v0.6.1-gd6044c25
> 2018-01-15 23:25:05 PothosFlow.MainWindow: Initialization complete
> 
> 2018-01-15 23:25:54 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libLogicAnalyzer.so): failed safe load
> 
> 2018-01-15 23:25:54 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libSpectrogram.so): failed safe load
> 
> 2018-01-15 23:25:54 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libConstellation.so): failed safe load
> 
> 2018-01-15 23:25:54 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libWaveMonitor.so): failed safe load
> 
> 2018-01-15 23:25:54 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libPeriodogram.so): failed safe load
> 
> 2018-01-15 23:25:54 Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(/usr/local/lib/Pothos/modules0.6/plotters/libQwtWidgets.so): failed safe load
> 
> 2018-01-15 23:25:54 ::1: ALSA lib pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
> 
> 2018-01-15 23:25:54 ::1: ALSA lib pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
> 
> 2018-01-15 23:25:54 ::1: ALSA lib pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
> 
> 2018-01-15 23:25:54 ::1: ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
> 
> 2018-01-15 23:25:54 ::1: ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
> 
> 2018-01-15 23:25:54 ::1: ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
> 
> 2018-01-15 23:25:54 ::1: ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
> 
> 2018-01-15 23:25:54 AudioSource: Using default through ALSA

so the corresponding plotter modules did not appear in GUI of pothos.

guruofquality, 2018-01-15:
This usually ends up being the result of multiple copies of qwt compiled against qt5 and qt4 installed. Can you see what installed on your system in terms of libqwt either from the package manager or /usr/local stuff?

Ubuntu for example used to have identical names for qwt compiled for qt4 and qt5, and it used to cause a lot of issues like this, now its named accordingly (qwt-qt5) and there are less issues, at least on ubuntu.

You can also remove every copy of qwt and rebuild pothos plotters, it will actually build an internal copy of qwt from a submodule -- if that helps.