pothosware/PothosCore #79: Problems building a C blocks

TheJimD, 2016-04-10:
Hi Josh,
As a first step, I am trying to create a simple block using C++ to add the data from two input ports and send it to an output port (similar to the Python example). I have some existing C code to decode a specific signal that I eventually want to turn into a block - I'm using the 2-input adder to learn the mechanics of creating a block.  I am using Windows 7 & Visual Studio 15.

I have been following the instructions in the Block Coding Guide and have hit a few snags.  I was able to use the Pothos_Module_Util & cmake to create a Visual Studio solution for my block.  I copied the structure of MyBlock.cpp, added my code into the work() function, and was able to compile my .cpp module.

When i execute the "cmake --build . --config RelWithDebInfo" from command prompt I get a LNK1112 error indicating "machine type x64 conflicts with target machine type x86".  I noticed that the link command includes both the /machine:X64 and /machine:x86 options.  I was able to go into the Visual studio property page under Linker options and manually change /machine:x86 to /machine:x64 in the "additional options" field.  When I did that I could build the .dll for my block from inside Visual Studio but cannot build it from the command line using the cmake command.

This error prevents me from running the final "cmake --build . --config RelWithDebInfo --target install" command.  I'm assuming that I need to do this final step to add my block to the Pothos environment. I tried just copying my .dll from the build/x64/RelWithDebinfo folder into the PothosSDR/lib/Pothos/Modules folder but that didn't make my block appear in the block tree.

What can I do to make the cmake command work and install my block into the environment?  Is this something in my Visual Studio setup (I did a basic install using the default settings) or something in the files that cmake & the Pothos_Module_Util generated?  Or should I just do the building & installing from inside Visual Studio?

I feel like I am close to being able to create my block but that I am getting hung up on the mechanics of the building process.  I can send you screen captures and/or files if that will help.

Thanks,
   Jim

guruofquality, 2016-04-10:
> "machine type x64 conflicts with target machine type x86"

When you generated the visual studio solution with cmake, you generally have to select the compiler. In this case I believe you selected the 32-bit version of Visual Studio. You should make sure that its the x64 one. If you were using cmake gui, there is usually a nice drop-down menu to select between the compilers.

Example (notice Win64):
[image]

Hope that helps

TheJimD, 2016-04-11:
That did help.  When I used the cmake GUI & selected Win64 I was able to build & install my block using the cmake commands.  My TestAdd.dll was copied to the PothosSDR/lib/Pothos/Modules/jim folder (I specified "/jim" in the Pothos_Module_Util Destination line of my cmakelists.txt).  So I think I have the "how to build & install" process sorted out now.  (CMake GUI is apparently my friend :) )

But I think I'm still missing one bit of vital magic. When I open the Pothos GUI, I don't see my block listed in the block tree.  What do I need to do to get my block to show up in the GUI?  Is this where I need to write a main() function to instantiate my block?

Thanks,
   Jim

guruofquality, 2016-04-11:
A few things you can check:

1) Is there a block description comment? https://github.com/pothosware/pothos/wiki/BlocksCodingGuide#block-desciptions

2) Does the cmake code have ENABLE_DOCS set in the call to POTHOS_MODULE_UTIL()? https://github.com/pothosware/pothos/wiki/BlocksCodingGuide#compilation-scripts

* more info here: https://github.com/pothosware/pothos/wiki/BlockDescriptionMarkup

3) If the module containing your block is loaded properly, you should be able to see it in the plugin-tree. So if you run the following command PothosUtil --plugin-tree=/blocks

You would see a path printed /blocks/path/to/my_block, and  /blocks/docs/path/to/my_block. Both should be in there

TheJimD, 2016-04-12:
Ok - that was another silly mistake on my part.  I didn't realize that I needed the block description comment section in order to load the block into the GUI.  In focusing on trying to create the code for my block, I glossed over that section figuring that I didn't need the comments just for a simple test component!

Once I got the |Factory line in the block description comments and the BlockRegistry line at the end of my block to match, I was able to see my block into the GUI block tree, add it to my flow diagram, and see it work correctly.  

So that was a good learning experience and now I can move on to creating a more complex block. I also want to experiment a little with the labels & messages passed between blocks - I think I am going to need them for what I want to implement.

Thanks again for your help!
  Jim D

guruofquality, 2016-04-13:
Always good to hear about success. Thanks!

closed 2016-04-13 by guruofquality