pothosware/PothosCore #82: Wav File Source block error

TheJimD, 2016-05-17:
Hi Josh,

I want to use the Wav File Source block as a source for testing a signal decoding block that I am working on.  When I load the Wav File Source block into the GUI, it displays an error.

[image]

Is this something that I need to download the source & rebuild?

For what it's worth, the Gnuradio Wav File Source block & my wav file work fine in GRC (based on the filename it looks like the Pothosware block is based on the Gnuradio block). 

On another topic I saw on RTL-SDR.com that you are involved with the Lime SDR project.  That is a really interesting looking device. I hope that works out for you.

Thanks for the help,
    Jim D

guruofquality, 2016-05-17:
@TheJimD I appreciate the report. It looks like this one is my bad.

It looks like the file source and sink blocks (including the wav ones) use "const char *" to specify the filename -- and there is no supported conversion between std::string which holds the filename and the destination type used here which is "const char *".

I debating whether or not to 1) add support for conversions between std::string and const char * or to 2) fix the gr wrapper to call the std::string's .c_str() method. In either case, I need to fix this and put out a new build.

TheJimD, 2016-05-18:
OK - I may try modifying my test signal generation code to write a binary file and use the binary file source until the wav file source is fixed.

Thanks,
   Jim

guruofquality, 2016-05-22:
@TheJimD Fixed in gr-pothos (the support for const char *):

* https://github.com/pothosware/gr-pothos/commit/f31757444234913d9d8e6fdc913b778866488008
* https://github.com/pothosware/gr-pothos/commit/b54dfd182f758d3301e73a802034903595dc7615

New installers "PothosSDR-2016.05.22-vc*" are available with this fix:
http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D

closed 2016-05-22 by guruofquality