pothosware/PothosCore #91: Dropping MSVC 2012 support: variadic templates and more

guruofquality, 2016-07-05:
**We ended up not using variadic templates** to support MSVC2012 which had incomplete c++11 support. Now that 2012 is dropped in favor of the more compliant 2013 and 2015, we should cleanup use of generated headers using variadic templates.

* [ ] All of the Callable, Proxy, ManagedClass APIs can be cleaned up. There is no need for this template generation system: https://github.com/pothosware/pothos/blob/master/include/CMakeLists.txt#L38

**Further, we avoided certain API calls** because it would require even more generation like this, so...

* [ ] Create a templated emit() API call on Pothos::Block to replace the use of callVoid(signalName, args...)
* [ ] Create a BlockRegistry::make(path, args...) factory call to neaten all of the BlockRegistry invocations that operate in the current process.

**Other misc VS2012 work arounds**:

* [ ] remove pothos_explicit ifdef in Config.hpp (use explicit)
* [ ] remove MathCompat.hpp, all of these functions were for 2012
* [ ] revert unit tests used list initialization for test vectors but were "fixed" for 2012

guruofquality, 2016-07-14:
Done! This does mean that many of the submodules master branches only compile with the master branch of pothos. But thats bounds to happen with API additions.

closed 2016-07-14 by guruofquality

gvanem, 2016-12-28:
These `comms` files:
```
waveform/NoiseSource.cpp
waveform/WaveformSource.cpp
digital/SymbolMapper.cpp
```
still seems to have this:
```
#include <Pothos/Util/MathCompat.hpp>
```

Aren't those DLLs supposed to be built? But for me, things compile on MSVC-2015 w/o this header.