pothosware/PothosCore #95: /proxy/python self test failure

mehdideveloper, 2016-07-11:
Hi
Running "PothosUtil --self-tests" returns 9 failed tests (out of 120)

Failed 9 out of 120 tests
FAIL: /proxy/python/tests/test_basic_types
FAIL: /proxy/python/tests/test_compare_to
FAIL: /proxy/python/tests/test_containers
FAIL: /proxy/python/tests/test_call_module
FAIL: /proxy/python/tests/test_serialization
FAIL: /proxy/python/tests/python_module_import
FAIL: /proxy/python/tests/test_python_module
FAIL: /proxy/python/tests/test_python_block
FAIL: /proxy/python/tests/test_signals_and_slots
They are all the same:

| Proxy Environment Factory Error: Pothos::ProxyEnvironment::make(python) |
| -> {PythonProxyEnvironment::findProxy(Pothos): dlopen(/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_io.so, 2): Symbol not found: __PyCodecInfo_GetIncrementalDecoder |
| Referenced from: /usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_io.so |
| Expected in: flat namespace

So it just fails all the "/proxy/python" tests.
Any ideas?
(I've tested "hash -r python" with no success)
Python is installed using brew (Python 2.7.11)

guruofquality, 2016-07-11:
Its possible that the app linked against the brew version of python but used the headers from the system install of python.  I would be curious to see the output of ``brew reinstall pothospython --verbose``, the cmake configure output will tell which includes and libs were selected. Its just a guess, but if thats the case I'm not sure how one normally differentiates between the two in homebrew or if the strategy should be to only use the brew python and somehow exclude the system one. On my tests I dont use python2 FWIW.

mehdideveloper, 2016-07-11:
Thanks. That was it.
I uninstalled pothos-python and built it from source. I changed the cmake configuration options by running "ccmake .." in the "build" directory and then pressed "t" to go to cmake's advanced mode and changed python paths to brew's python.
Now all the tests return success.

closed 2016-07-11 by mehdideveloper

guruofquality, 2016-07-11:
Just FYI, I found some notes on this: https://github.com/Homebrew/brew/blob/master/share/doc/homebrew/Homebrew-and-Python.md#homebrew-provided-python-bindings It may be the case that the recipe for the cmake stuff is not properly selecting python the way it should. If python and python-config from brew are both the ones in the PATH, then it should use homebrew python only.