pothosware/PothosCore #106: Intel PROSet/Wireless Tools prevent grc from working

Hoernchen, 2016-11-28:
GRC will not start, the helper script will tell you that IMPORT_GTK failed to load a DLL.
Three of my students ran into this issue, the fix is to purge the PATH, because the Intel PROSet Wireless software will add itself to the global system PATH, and it appears to include a dll (apparently zlib1.dll) that will be loaded by gtk instead of the one supplied by the gtk runtime.
This is also a reminder that adding the GTK runtime supplied by Pothos/GRC to the global system PATH is probably not a good idea...

guruofquality, 2016-11-30:
Theres always a bit more work involved with the GRC dlls and sometimes there is DLL hell. So, I have seen a similar issue with the gtk dlls in regards to the bundled GRC before. My suggested fix for now is to use a tool like rapid environment editor (which makes it easier to edit paths) and move the GTK dll path higher up in priority. If it appears earlier in the list, then the correct zlib1.dll will take precedence.

Typically installers set paths for command line use. So its unfortunate to rely on the PATH for gui launchers. I think that usually, start menu short cuts specify the start-in to get certain DLLs in the PATH with precedence, or maybe have a launcher bat which sets the PATH...

I might have to come up with something more creative. I made a custom launcher for GRC to handle the python dependencies and other things: https://github.com/pothosware/gnuradio-companion-exe Maybe it could do more when loading grc so we dont have to rely on the PATH variable.

guruofquality, 2016-12-05:
I added the following change to the gnuradio companion launcher. It sets the environment variables which should not be needed for most because of the installer, however, it should help with DLL path priority and the problem described above: https://github.com/pothosware/gnuradio-companion-exe/commit/330ce3cd9bc3761f7294dde1197ea69e819c5070 Also, I uploaded new installers, but I made this change afterwards... so it will be in the next one (just FYI).

closed 2016-12-05 by guruofquality