pothosware/PothosCore #110: cannot find SoapySDR library in /home/lib

chakpongchung, 2017-01-26:
cpchung:build$ cmake ..
-- Poco_VERSION: 1.7.2
-- Poco_INCLUDE_DIRS: 
-- Poco_LIBRARIES: Poco::Foundation;Poco::Util;Poco::JSON;Poco::XML;Poco::Net
-- Build type not specified: defaulting to release.
-- muparserx_VERSION: 4.0.7
-- muparserx_INCLUDE_DIRS: /usr/include/muparserx
-- muparserx_LIBRARIES: /usr/lib/x86_64-linux-gnu/libmuparserx.so
-- Spuce_VERSION: 0.4.3
-- Spuce_INCLUDE_DIRS: /usr/include
-- Spuce_LIBRARIES: /usr/lib/x86_64-linux-gnu/libspuce.so
-- PYTHON_EXECUTABLE: /home/cpchung/anaconda2/bin/python
-- POTHOS_PYTHON_DIR: ${prefix}/lib/python2.7/site-packages
-- PYTHONLIBS_FOUND: TRUE - 2.7.12
-- PYTHON_INCLUDE_DIRS: /home/cpchung/anaconda2/include/python2.7
-- PYTHON_LIBRARIES: /home/cpchung/anaconda2/lib/libpython2.7.so
CMake Error at /home/cpchung/SoapySDR/cmake/SoapySDRConfig.cmake:161 (message):
  cannot find SoapySDR library in /home/lib
Call Stack (most recent call first):
  sdr/CMakeLists.txt:16 (find_package)

guruofquality, 2017-01-26:
@chakpongchung Its looks like this happens when the repositories are cloned in the home directory. Probably something to do with cmake's default search path for project configs. I didnt expect this behaviour, but apparently it just innocently searching $home/project/cmake. Problem is it finds the source directory and not the installed version of the files.

This wasnt the intended behaviour, but the work around for now is either put the git clones in a subdirectory, or pass `-DSoapySDR_DIR=/usr/local/share/cmake/SoapySDR/` to pothos cmake command.

guruofquality, 2017-06-02:
Fixed for SoapySDR - https://github.com/pothosware/SoapySDR/commit/cd4e0344f36a1a1a8ae2e2871bced2a9228cbe5b

Will also do this for pothos project config for a similar reason

guruofquality, 2017-06-02:
Also for pothos: https://github.com/pothosware/pothos/commit/748742ccc0685c66d728cc6c1ddf198aef1b6e1a

closed 2017-06-02 by guruofquality