pothosware/PothosCore #114: SDRBlock:activate failure.

romeojulietthotel, 2017-04-10:
Built from git tag pothos-0.4.2 on Linux
All 121 tests passed!

But when I Activate Topology this is what happens:

```
PothosGui.TopologyEval.commit ⇒ 
Framework Topology Connect Error: Pothos::Topology::commit(): Framework Topology Connect Error: SDRSource0.activate(): Exception: virtual void SDRBlock::activate() - stream not setup!

```

Device Args: {"driver":"lime"}

I'm just trying the pothos-demos at this point.

Really have tried everything I could think including strace'ing it and rebuilding the software many times.

I also built everything on git master with same results. All my other software is working fine (Gqrx, CubicSDR, GRC, rtl_sdr).

romeojulietthotel, 2017-04-10:
_(Maybe I have filed this issue against the wrong product? Should I have filed against pothos-gui?)_

I have noticed what might be the problem. PothosGui->Host Explorer->System Info->SoapySDR info->Factories = null

But 

`SoapySDRUtil --info`

shows me:

`Available factories...lime, null, `

### Note that PothosGui is not showing a "lime" factory.

Is that my problem? If so how do I remedy? If not then is it possible that there's a conflict? For example in PothosGui I have two ways to select the SDR source:

1. Block Tree->SDR->SDR Source
2. Block Tree->Sources->SDR Source

If none of this matters then I am out of ideas @guruofquality.

romeojulietthotel, 2017-04-10:
Aha!  I think I've figured it out. Would still like your input whenever you can spare the time.

I had to set:

`SOAPY_SDR_PLUGIN_PATH to point to limesuite/lib64/SoapySDR/modules0.6-dev`

i.e. export SOAPY_SDR_PLUGIN_PATH

Is this an rpath setting?  I am not that familiar with cmake so I will look over the cmake log.

Funny how sometimes just going through the motions and filing an issue will lead me to a solution.  Happens a lot...  But maybe not all for nothing if it helps someone else somewhen later.

guruofquality, 2017-04-10:
> Aha! I think I've figured it out. Would still like your input whenever you can spare the time.
I had to set:
SOAPY_SDR_PLUGIN_PATH to point to limesuite/lib64/SoapySDR/modules0.6-dev
i.e. export SOAPY_SDR_PLUGIN_PATH

Can you share some of the LIB_SUFFIX and CMAKE_INSTALL_PREFIX that were used for lets say, limesuite, soapysdr, or pothos?

In general, SoapySDR is going to find modules that are in the same installation prefix, but `limesuite/lib64` seems unexpected.

> Note that PothosGui is not showing a "lime" factory.

Not sure why that would be. Is there a chance there are multiple copies of some of the software, like one from source, and one from a package manager?

> Maybe I have filed this issue against the wrong product? Should I have filed against pothos-gui?

pothos/issues is probably a good catch all, dont worry about it

romeojulietthotel, 2017-04-13:
Didn't set the LIB_SUFFIX IIRC and CMAKE_INSTALL_PREFIX was different for each app/lib.

The "lime" factory not showing due to SOAPY_SDR_PLUGIN_PATH not set.

And no duplicate packages, etc. All built from source.

I have found that if one deviates at all from standard locations for any of the software in the large SDR software population it requires lots of cmake engagement.  Cmake's good but seems like every package that uses it only works for a very few, narrow cases.

guruofquality, 2017-04-15:
> I have found that if one deviates at all from standard locations for any of the software in the large SDR software population it requires lots of cmake engagement. Cmake's good but seems like every package that uses it only works for a very few, narrow cases.

Thats probably true. Its a matter of finding dependencies usually, and that can be quite manual for non standard paths. If you happen to use the same CMAKE_INSTALL_PREFIX for an entire set of related software + cmake projects, its usually pretty well-behaved.

romeojulietthotel, 2017-04-16:
To keep with the latest code across all the moving parts I tend to stay with most current code.  Unfortunately distro maintainers don't keep up since the user base for SDR is a small subset of the entire userbase.

But to install everything in the same path causes significant pain if some problem arises so I keep it separate.

I will close this now since it's specific to my environment. At least I am now learning cmake. Apologies for extra noise.

closed 2017-04-16 by romeojulietthotel