pothosware/PothosCore #116: Pothos not cleaning up procs and subprocs.

romeojulietthotel, 2017-04-14:
Just discovered that there were four PothosUtil and two PothosGui procs running even though I had exited PothosGui.  Two of the PothosUtil procs were marked `<defunct>`.

This causes lots of problems if it's not known to the user that there are still procs running (well I don't know what they were doing). If I see this again I will try to gather some more info. This is all I have now.

All built from tag pothos-0.4.2 on Linux.

romeojulietthotel, 2017-04-16:
I need to test this but something I just thought of....maybe this problem occurs because I launch pothos from a .sh script and the script uses the full path to launch the gui i.e. /path/to/pothos/bin/PathosGui

romeojulietthotel, 2017-04-19:
If there's no device connected and pothos has a few topologies to load and then I exit it seems to show this problem. 

The PothosGui began consuming memory as well and I eventually killed the procs.

romeojulietthotel, 2017-04-19:
Not sure if this is a feature but I notice that if pothos starts and I have a few topologies saved that pothos looks at them and does "stuff".  Sometimes looping over them like in a race condition.

Pothos seems to step on itself,  i.e. one thread can access the hardware and another cannot for example.

guruofquality, 2017-04-20:
@romeojulietthotel thanks for reporting

I think you have stumbled on two issues. I think there is some kind of race or cleanup issue in the GUI but I'm not sure as to the reason. I think I have seen it happen even when a tab is closed, so its not strictly the GUI shutting down. Its somewhat random, but something locks up when shutting down, maybe memory is being stepped on. I have had a hard time tracking down precisely the issue.

> Pothos seems to step on itself, i.e. one thread can access the hardware and another cannot for example.

Thats true, the process has one handle open to a particular device, it cant be used in another process. This could lead to one open design claiming a device that another might use. A quick fix might involve disabling the block when not using it (short cut is "d"). But automatically not maintaining active blocks in the memory for tabs that are not selected/in-focus may be a possible solution.

romeojulietthotel, 2017-04-20:
Disabling is fine but if I have three tabs, to disable I have to switch to each tab->Select all->disable. Maybe this is nitpicking. I am conditioned to have the app start and do nothing until I activate it. For example Gqrx, URH, Gnuradio, etc.

guruofquality, 2017-05-03:
This issue was moved to pothosware/pothos-gui#146

closed 2017-05-03 by guruofquality