pothosware/PothosCore #117: Toolkit renaming PothosSoapy, PothosFlow, and etc

guruofquality, 2017-04-23:
Were looking to [tag everything](https://github.com/pothosware/pothos/issues/115) and make proper releases out of the master branch. After that point, I'm considering various project renaming for the sake of convention, consistency, and preference.

**pothos-blocks -> PothosBlocks and the like**

While the war between camel case and underscores may rage on, this is a camel case project, so is SoapySDR, and all of the SoapySDR repos themselves. And so the various toolkits will be named accordingly. This is only a github project rename and not a code change. The url redirection should ensure that nothing breaks in terms of links, and new links will be updated in the wikis and repos.

**pothos-sdr -> PothosSoapy**

Few reasons here, pothos-sdr ended up being ambiguous with the PothosSDR windows installer (my bad), pothos-sdr is not the only toolkit for SDR devices (example IIO), and many requested the name gr-soapy over the name gr-sdr for the gnuradio equivalent project. Although the blocks will be renamed, but aliases will be maintained as to not break any existing designs.

**pothos-gui -> PothosFlow**

Calling the GUI literally gui, is quite accurate, but also very lame. Its like naming your dog, "dog". Qt has graphical applications like QtCreator, QtDesigner. I think the designer gui could use some kind of moniker as well. Since we are designing flows of interconnected processing blocks, I thought PothosFlow might be a nice, short, and to the point kind of name. The only breakage here is that the executable name will change.

Anyway, this is by no means finalized, so I'm happy to hear about suggestions.

romeojulietthotel, 2017-05-28:
I like the idea of renaming. Yes the gui part is lame but at least you used the acronym and not the full spelling, i.e. GraphicalUserInterface. ;)  My first visit to this project left me confused by all the different parts. Not completely un-confused yet actually.  But that seems to be the case across the SDR software spectrum.   

I never understood the camel-case mini-drama.

FiachAntaw, 2017-05-30:
This also sounds good to me. I agree with the change of [pothos-sdr](https://github.com/pothosware/pothos-sdr) to PothosSoapy, the first time I saw it I assumed it was the same as [PothosSDR](https://github.com/pothosware/PothosSDR) and was confused when I saw both were present.

guruofquality, 2017-09-09:
Work completed! Please let me know about any broken/stale links or out of date wikis.

* Users following master branch will need to `git submodule update --init` after pulling.
* Also delete any old modules to avoid clashing `rm -rf /usr/local/lib/Pothos/modules*`
* Also keep in mind PothosGui executable is now PothosFlow, so you have to run that instead.

closed 2017-09-09 by guruofquality