pothosware/PothosCore #118: Ubuntu documentation is a bit of a mess

kjcole, 2017-04-30:
I'm running Ubuntu 16.04 LTS.

Following the install instructions at https://github.com/pothosware/pothos/wiki/Ubuntu leads the user on a merry chase.

It says to install **pothos-all**, but it depends on **pothos** which is not going to be installed. Installing **pothos** manually and then trying to install **pothos-all** to pick up the missing pieces tries to uninstlall the dependency it needs. However, selecting it and deselecting it (after it's marked all the packages it will need) appears to work.

Moving on to the **python-pothos** and **python3-pothos** installs, they first want to uninstall the newer **libpothos0.5 pothos** in favor of **libpothos0.4-3**, (as well as **pothos** -- the undependable dependency of **pothos-all**, and **pothos-python-dev**). **Python3-pothos** then wants to uninstall the just barely installed **pothos0.4-3-modules-python** and **python-pothos**, in favor of the Python3 versions.

Eventually, we get to the test by running **PothosGui**... which hasn't been installed at this point. Installing that and running it leads to:

`Exception: Assertion violation: _readfd != -1 in file "/build/poco-GfkNHM/poco-1.7.2/Foundation/src/PipeImpl_POSIX.cpp", line 66`

guruofquality, 2017-04-30:
I am in the middle of uploading packages to the PPA as we speak. Thats probably the cause.

kjcole, 2017-04-30:
Ah. Thanks. Bad timing on my part then.  Meanwhile I started going down the "install from source" road on https://github.com/pothosware/pothos/wiki/BuildGuide#ubuntu and was met with:

```
$ sudo add-apt-repository -y ppa:pothosware/support
Cannot add PPA: 'ppa:~pothosware/ubuntu/support'.
ERROR: '~pothosware' user or team does not exist.

$ sudo add-apt-repository -y ppa:myriadrf/drivers
Cannot add PPA: 'ppa:~myriadrf/ubuntu/drivers'.
ERROR: '~myriadrf' user or team does not exist.
```

but I'll try again later.

guruofquality, 2017-04-30:
The PPA should be published now. I updated the soapysdr, poco, pothos, modules, etc. A lot of library and ABI changes, so all of the software had to be rebuilt. It should be ok now. I will be testing the zesty debs soon to see if anything comes up.

As far as the `Cannot add PPA` error. I havent seen that before, and I dont think the upgrade of packages could cause it. Thats the recommended command on the launchpad site and it worked for me earlier. I wonder if ubuntu was just down for the count (bad timing again).

kjcole, 2017-04-30:
On Sun, Apr 30, 2017 at 2:59 PM, Josh Blum <notifications@github.com> wrote:

> The PPA should be published now. I updated the soapysdr, poco, pothos,
> modules, etc. A lot of library and ABI changes, so all of the software had
> to be rebuilt. It should be ok now. I will be [...]

​Yeah. I deleted the PPA error comment when I realized that only moments --
well perhaps an hour -- earlier, the very same PPAs had worked. So, I too
assumed a temporary glitch.  (Not at that computer now. So, I can't retry
the package installs.)
​
Physical
[image: [Photo: Kevin Cole]]
Kevin Cole, RHCE <http://launchpad.net/~kjcole/>
Team Contact
Ubuntu Linux DC "LoCo" <http://dc.ubuntu-us.org/>
Washington, DC (US)
Virtual
E-mail: kjcole (at) ubuntu•com
WWW: http://dc.ubuntu-us.org/
GPG Key ID: 0x3E696927 <http://research.gallaudet.edu/~kjcole/pubkey.php>

kjcole, 2017-05-01:
Yea verily, this was the trouble. Once the latest and greatest packages were in the PPA, all went well. I now have a runnig PothosGUI. Don't yet have a clue what to do with it, but it's making static and showing graphs. Yay! ;-)

closed 2017-05-01 by kjcole