pothosware/PothosCore #119: PothosUtil: error while loading shared libraries: libPothos.so.0.5: cannot open shared object file: No such file or directory

Kishaihd, 2017-05-02:
I looked through the issues and did some google searching before posting this issue.
When I try to run PothosGUI or PothosUtil, I get the message:
`PothosUtil: error while loading shared libraries: libPothos.so.0.5: cannot open shared object file: No such file or directory` 
That file is located in /usr/local/lib, but I don't know where Pothos is looking, nor do I know how to tell it where to look. 

Has anyone else had this problem? I feel like it should be an easy fix, it's just not something I've encountered before.

guruofquality, 2017-05-02:
Its just a generic shared library location issue for your OS. If this is debian or ubuntu, you need to run `sudo ldconfig` after installing libraries. There is also an option of setting LD_LIBRARY_PATH to contain /usr/local/lib.

guruofquality, 2017-05-09:
Cool, let us know if that doesnt work

closed 2017-05-09 by guruofquality

Kishaihd, 2017-05-09:
Oh hey, so I'm running arch Linux; to fix it I added the library path to LD_LIBRARY_PATH (which was empty prior to adding /usr/local/lib), as you suggested. 

Thanks!