pothosware/PothosCore #120: Apparent race condition in Arithmetic Block / Read-Before-Write Inlining bug?

FiachAntaw, 2017-05-08:
[This topology](https://github.com/pothosware/pothos/files/982684/rbw-bug.pothos.txt) produces the following output when run on Pothos 0.5.0:

[image]

It appears that occasionally the Arithmetic block processes the buffer on port 0 in-place before it has been rendered by the Wave Monitor block, causing the data displayed on the Wave Monitor block to be displaced by 1024.

The issue remains when the Pacer block is removed, and also occurs if the Wave Monitor block is replaced by a Binary File Sink block and subsequently replayed directly into a Wave Monitor Block. It occurs for integer, float and complex float dtypes. I have not observed this occurring when input ports 0 and 1 of the Arithmetic block are swapped.

Changing line 301 of WorkerActor.cpp from:
```c++
        if (tryRBW and port._buffer.useCount() == 2) //2 -> accumulator + this port
```
to:
```c++
        if (false) //2 -> accumulator + this port
```
appears to stop the issue from occurring, but of course also disables the read-before-write buffer inlining optimization.

guruofquality, 2017-05-09:
Thanks for the reporting. I was able to replicate this. It does really look like the buffer is getting inlined when its technically still held as a reference by another block. The wave source should be passing a reference (incrementing) to both the adder and the pacer, before discarding its own reference (decrementing). So while I dont think its a race on the reference count, it could be an accounting issue where we drop the reference before we should. I'm looking into it.

guruofquality, 2017-06-05:
Solution: https://github.com/pothosware/pothos/commit/6dbb6118bbf25213da3544f7f3702c20fe9748bb

closed 2017-06-05 by guruofquality