pothosware/PothosCore #125: PothosUtil "Application: Exception"

masmirnoff, 2017-05-25:
During the build of the library (ArchLinux, gcc 7.1.1), I get the following parser error for all files containing documentation blocks (PothosDoc):

> Syntax error: BlockDescriptionParser(/home/redcat/workspace/lib/build/pothos/blocks/file/BinaryFileSource.cpp)
>  -> {Inconsistent indentation: 25: *}

If you fix the line "_*" with "_*_" (_ means a space) in the documentation block, then the parser falls with the following error:

> Syntax error: BlockDescriptionParser(/home/redcat/workspace/lib/build/pothos/blocks/file/BinaryFileSource.cpp)
>  -> {Expected |widget SpinBox(args...): 33: * |widget DTypeChooser(float=1,cfloat=1,int=1,cint=1,uint=1,cuint=1,dim=1)}

What could be the problem?

guruofquality, 2017-05-25:
OOO, thats pernicious. I bet that something is subtly different with either the std::string functions or the version of poco thats installed (regexp stuff from poco).

I will leave this here since I cant debug it now. Something is off probably in one of these two functions:

* Either this function which parses the comment block: https://github.com/pothosware/pothos/blob/master/lib/Util/BlockDescription.cpp#L213
* Or the one before it which extracts the comment block from the rest of the file: https://github.com/pothosware/pothos/blob/master/lib/Util/BlockDescription.cpp#L129

So I would add prints and see what the function thinks its getting as input. You can run the app from the build dir like so: `apps/PothosUtil --doc-parse=path_to_file_block.cpp`

masmirnoff, 2017-05-25:
I tried to not use the system poco library, but from your repository, the effect is the same.

masmirnoff, 2017-05-25:
I added the following code (`std::out`) to the function `static std::vector<CodeBlock> extractContiguousBlocks(std::istream &is)`

```
        if (not partial.empty()) line = partial;
        else
        {
            lineNo++; //starts at 1
            std::getline(is, line);
            std::cout << "extractContiguousBlocks:147 -> line is: \"" << line << "\"" << std::endl;
        }
```

and to the function static `Poco::JSON::Object::Ptr parseCommentBlockForMarkup(const CodeBlock &commentBlock)`

```
    //search for the markup begin tag and record the indent
    for (const auto &codeLine : commentBlock)
    {
        std::cout << "parseCommentBlockForMarkup:237 -> codeLine is: \"" << codeLine.toString() << "\"" << std::endl;
        std::string line = codeLine.text;
        Poco::RegularExpression::MatchVec matches;
```

And got the following output for FileBlocksDocs.cpp
[output.txt](https://github.com/pothosware/pothos/files/1028379/output.txt)

There are some problems, for example characters "moved from the end to the beginning of lines:

> "xtractContiguousBlocks:147 -> line is: "// Copyright (c) 2014-2016 Josh Blum
> "xtractContiguousBlocks:147 -> line is: "// SPDX-License-Identifier: BSL-1.0
> "xtractContiguousBlocks:147 -> line is: "
> "xtractContiguousBlocks:147 -> line is: "#include <Pothos/Framework.hpp>

masmirnoff, 2017-05-25:
I am add test of the last symbol in `line` for equality to '\r'? and remove it. After that the parser began to work normally.

in function `static std::vector<CodeBlock> extractContiguousBlocks(std::istream &is)`

```
         if (not partial.empty()) line = partial;
        else
        {
            lineNo++; //starts at 1
            std::getline(is, line);
            if(line.back() == '\r') line.pop_back();
        }
```

guruofquality, 2017-05-25:
http://www.cplusplus.com/forum/general/51349/

> If you are using a program under Linux but reading a file with a Windows newline sequence, you will get those '\r's at the end of the string.

I should probably make sure that the line is stripped of newline and carriage return characters after the std::getline call. It should be robust. But out of curiosity, is it possible that you cloned the repo with windows newlines (\r\n) somehow?

guruofquality, 2017-05-27:
in any case, thanks for the patch and investigation: https://github.com/pothosware/pothos/commit/929f7e2ef85593d6c588c5e91876c8fccfac627f

closed 2017-05-27 by guruofquality