pothosware/PothosCore #131: update to modern use of cmake (> 3.3)

guruofquality, 2017-10-10:
Mostly we wanted to support OS releases without updating cmake, but that time is coming where its safe to put those in the past. Modern cmake usage supports some features that vastly simplify its use:

* targets can have associated defines, public/private libs and includes
 * embedded properties like this mean that variables done have to be passed around, targets are merely linked together to bring in includes, defines, private libs 
* support for exporting and importing project configs and targets
* remove ifdefs for older unsupported features/deprecated calls
* new syntax for compiler feature support like c++11/14

Probably look at doing this for PothosCore and SoapySDR after 0.6 release. Hopefully the builder macros stay mostly the same, so plugins dont need to change, but users can take advantage of the new target properties when it makes sense to do so.

ncorgan, 2020-04-05:
Will be part of: https://github.com/pothosware/PothosCore/tree/modernize

guruofquality, 2020-04-05:
Im def tagging this repo before I merge this one

ncorgan, 2020-04-05:
Certainly. This is definitely meant for 0.7 master.

ncorgan, 2020-04-05:
I may backport my commit to override the C++ standard, but default behavior wouldn't change.

closed 2021-07-05 by guruofquality