pothosware/PothosCore #151: Application error at PothosGui startup

bondarevky, 2017-12-21:
Application can't start up and displays the following message:
[image]

Windows 10 Pro 1703

guruofquality, 2017-12-21:
Any chance the wifi connection is down? I remember that caused some socket  and or dns connection to fail, even though it was just localhost. It just couldnt resolve anything with the windows network config as it was. Just a hunch though: https://groups.google.com/forum/#!topic/pothos-users/HJhjCopYQ6I

bondarevky, 2017-12-21:
I tried to start PothosGui with only Wi-Fi connection, only Ethernet connection and with disabling all network adapters - no result. All connection settings are by default.

guruofquality, 2017-12-21:
This is probably a bug in poco (one of the dependency libraries) where its dying in an assert rather than throwing. So rather than a graceful error message, or even recovery, I get this process abort out of no where -- so I know thats not very helpful. And if we do track this down, poco probably needs an upstream patch.

Anyway: my second guess is that ipv6 is somehow disabled, and thats also causing it to fail what I assume is some kind of dns lookup for localhost or `::1`. The next step I would try is to click debug if visual studio was installed and to get the traceback. I dont know if you have that capability.

When I get the chance I will see if any of my other PCs are capable of reproducing this with the latest installer...

bondarevky, 2017-12-21:
Problem fixed by starting application with administrator privileges. Maybe there is some kind of permissions mismatch.

guruofquality, 2018-02-18:
This error comes out of poco when the process fails to launch, and we try to read the error stream. Replaced it with something more obvious: https://github.com/pothosware/PothosCore/commit/bd154850dfbf7d5a035700fe73d3325cb63857c3 Specifically for this bug, the windows permissions must have blocked the process from being launched if that make sense.