pothosware/PothosCore #153: compiled soapysdr, while making pothoscore, got libaudiosupport error

lrmodesgh, 2018-01-14:
last few lines of cmake..

```
/usr/bin/ld: cannot find -lasound
collect2: error: ld returned 1 exit status
make[2]: *** [audio/CMakeFiles/AudioSupport.dir/build.make:217: audio/libAudioSupport.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:4496: audio/CMakeFiles/AudioSupport.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 86%] Built target ControlWidgets
[ 86%] Built target Constellation
[ 86%] Built target LogicAnalyzer
make: *** [Makefile:163: all] Error 2

```
what am i doing wrong here ?

guruofquality, 2018-01-14:
Basically, cmake things it found portaudio development files courtesy of [FindPortaudio.cmake](https://github.com/pothosware/PothosAudio/blob/master/FindPortaudio.cmake). Essentially this runs `pkg-config --libs portaudio-2.0` to get a list of libraries to link against. For ubuntu, this happens to give me `-lportaudio -lasound -lm -lpthread`, so `-lasound ` also shows up for me at least (aka /usr/lib/x86_64-linux-gnu/libasound.so).

I don't think you did anything wrong, its just strange that portaudio has pkg config files that require linking to libasound, but its not present on the system. Perhaps a system os issue?

At least speaking for ubuntu ,this development file (/usr/lib/x86_64-linux-gnu/libasound.so) can be installed with `sudo apt-get install libasound2-dev`. Can you try something like that?

lrmodesgh, 2018-01-15:
thats a relief, i actually do searched for the libasound.so file and it was installed in the usual system lib directory, but was named as libasound.so.2. after making a link file with libasound.so it seems pothoscore make completed (still make shows only 86%). Then i proceeded for 

sudo make install

pothos installed finally. yay.. :)

lrmodesgh, 2018-01-15:
actually i tried in this relatively new distro. **solus**. so far pothos works in solus. Compilation from source is challenging, but also fun, interesting and learned stuff & pure.

closed 2018-01-15 by lrmodesgh

guruofquality, 2018-01-15:
> but was named as libasound.so.2

Typically you will see something like this, its 3 files:

```
lrwxrwxrwx 1 root root      18 Feb 13  2017 /usr/lib/x86_64-linux-gnu/libasound.so -> libasound.so.2.0.0
lrwxrwxrwx 1 root root      18 Feb 13  2017 /usr/lib/x86_64-linux-gnu/libasound.so.2 -> libasound.so.2.0.0
-rw-r--r-- 1 root root 1079592 Feb 13  2017 /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0
```

An actual so file with the full version number (libasound.so.2.0.0), a symlink for the ABI (libasound.so.2, this is what the library loader should be looking for), and a symlink for development with no version number (libasound.so). So I would say you are missing the development symlink. So thats something to look into or report to the package maintainer. But making the symlink manually should be fine of course.