pothosware/PothosCore #157: Can "PothosCore" be used for image processing?

DaeHeeLim, 2018-03-16:
Can "pothos" be used for image processing?

No related examples were found.

For Example, how do I pass an object such as cv::Mat in openCV?

guruofquality, 2018-03-16:
The message capabilities for input and output ports are probably best suited here: 

* https://github.com/pothosware/PothosCore/wiki/BlocksCodingGuide#posting-output-messages
* https://github.com/pothosware/PothosCore/wiki/BlocksCodingGuide#reading-input-messages

Example work()
```
auto msg = inputPort0->popMessage();
const auto &inMatrix = msg.extract<cv::Mat>();

cv::Mat outMatrix;
//TODO something with input and output matrix

outputPort0->postMessage(std::move(outMatrix));

```

DaeHeeLim, 2018-03-17:
thank you, i’ll try it

closed 2018-03-20 by DaeHeeLim