pothosware/PothosCore #161: Embedded platform with DSP cores

ferasboulala, 2018-09-26:
Let's pretend that I own an embedded platform with specialized DSP cores and I would like Pothos to run some of its processes on those said cores.

If I decide to write a topology from a PC and to remotely conduct my experiments with most of the processing done on the embedded platform, does the PC side of Pothos need to know how to call the DSP cores (ie. does it need to know the existence of the API that calls those cores) ? and do I have to install the entire Pothos on the embedded platform for this to work ?

guruofquality, 2018-09-26:
You would need PothosCore, PothosBlocks (probably for the network souce and sink), and whatever wraps the DSP module on the embedded side. No graphics or toolkits for anything you dont want to process. This assumes the embedded environment is a linux OS though.

ferasboulala, 2018-10-31:
Hi there again,

I am in the process of integrating Pothos with an embedded platform with DSP cores. To do so, I will have to either install Pothos on both a PC but I have a choice to either install Pothos on the device (it runs a minimal Linux) and modify it to work with DSP cores (blocks will call another program that manages DSP cores) or to mimic the Pothos proxy so it may communicate with the frontend on the PC.

Which option is more thesible in terms of workload mainly ? I have tried to make my way through the code but I do not seem to grasp everything about the Proxy so far.

guruofquality, 2018-11-05:
I would try to make the DSP blocks that manage the DSP core. If you cant do that, its probably easier to just stream samples over UDP rather than replicate the entire proxy thing. There is a UDP datagram block for that FWIW.

closed 2018-12-09 by guruofquality