pothosware/PothosCore #167: Ubuntu PPAs out of date

dracode, 2018-12-21:
Not sure if this is the right place for this or not...

I installed SoapySDR on Ubuntu Xenial today using the PPA listed in the documentation here.  Unfortunately, it does not work with BladeRF currently.

It looks like Nuand has updated libbladerf a few times since 2017-12-25 when the Pothos PPAs were updated and has made it to version 2.  So now things are out of sync.

I think this MAY be the issue?  Soapy has a dependency on libbladerf1 and bladerf, and bladerf wants to use libbladerf2...  which installs the firmware for v2.  which isn't backwards-compatible to libbladerf1.

Anyway, this is what happens when I run  SoapySDRUtil --probe with a BladeRF attached:
...
[INFO @ "/build/bladerf-zjSF8E/bladerf-2017.12-rc1/host/libraries/libbladeRF/src/version_compat.c":127] FPGA version (v0.9.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.

guruofquality, 2018-12-21:
You are totally right, its something I put off because its a big update. I just started this thread on myriadrf: https://discourse.myriadrf.org/t/updating-the-ppa-to-soapysdr-0-7/3892

Once the new soapy devel files are up, I will do the same for my pothos PPA as well.

Also new bladerf v2 support is tagged now in a release, just needs to be uploaded.

guruofquality, 2018-12-24:
FWI, soapysdr and soapybladerf should be updated, working on PothosCore and other projects that depend on it now.

guruofquality, 2018-12-30:
Should up updated, xenial, bionic, cosmic. Also arm64 arch too

closed 2018-12-30 by guruofquality