pothosware/PothosCore #169: Sugestion to scale some Label data when decimating or interpreting

ashley-b, 2019-03-30:
I was thinking it would be useful if rate transforming blocks i.e. decimating/interpolating filter could also scale some label data like "rxRate" and perhaps other user labels.
This could be done though a registry of labels to scale or a flag in the label maybe?

https://github.com/pothosware/PothosComms/blob/61d10f4d2c0523906207e91ec02ac86f6658653a/filter/FIRFilter.cpp#L307-L314

guruofquality, 2019-05-26:
I havent thought deeply about making some tag registry, but I did make this specific FIR block adjust the rate when passed in with the rxRate id specifically :-)

ashley-b, 2019-05-29:
Great, thanks for that. I will checkout it out.

I was thinking after I posted this sugestion that perhaps "rxRate" Label value should be transform in Label::toAdjusted method, so all new rate transforming blocks get this for free. But I have noticed its also used to shift labels due to data type transformation not just rate transforms.

As for a way to signal if a Label value should be adjusted, maybe a flag in the Label class would be the easiest way. So that a block adding labels (e.g SDR source) can set it, if it needs the value to be adjusted downstream.

Thanks again.