pothosware/PothosCore #170: Pothos Python Error

ChristoRei, 2019-04-12:
Hello,
i currently have problems using Pothos. My problem is that I get the following error messages:

I am aware that this error is known. But I already installed Python 3.6 and Numpy and added Python3.6 to the path. Unfortunately this didn't work. 

I would like to use blocks programmed with Python. Unfortunately the "Quadrature Demod" block I have from GNU Radio Companion shows me an error message. This is about "Callable Argument Error: Pothos failed to convert arg0 Object Convert Error".

I'd be very happy for help.

Kind Regards,

guruofquality, 2019-04-12:
> i currently have problems using Pothos. My problem is that I get the following error messages:
I am aware that this error is known. But I already installed Python 3.6 and Numpy and added Python3.6 to the path. Unfortunately this didn't work. I would like to use blocks programmed with Python. 

What error is a known error, how is python or numpy involved? Are you making custom blocks in python with https://github.com/pothosware/PothosPython/wiki

> Unfortunately the "Quadrature Demod" block I have from GNU Radio Companion shows me an error message. This is about "Callable Argument Error: Pothos failed to convert arg0 Object Convert Error".

I looked at the block. So python isnt a dependency here for runtime for what its worth. The .grc file for quad demod comes with an example formula that has been copied over `"samp_rate/(2*math.pi*fsk_deviation_hz/8.0)"`, but there is no definition for sample rate for deviation, so the error is expected. My guess is that you just have to fill in the gain.

ChristoRei, 2019-04-12:
Ah ok, thank you. 
No i actually dont want to make custom blocks, i just want to use all blocks available.
I thougth the error with the "Quadrature Demod" probably comes from the Python error.
I will check if setting gain for this Block helps.

Known Error: https://discourse.myriadrf.org/t/popthossdr-error-indicating-python-error/1636/12

There are lots of duscussions about this Error. Thats why i said its a known error...

I forgot to post the exact Error-Line in my first post. After the ":" i wanted to put the Error Line in, but i forgot. I will edit my post on Monday...

Thank you for your help

guruofquality, 2019-04-13:
> *Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(C:\Program Files\PothosSDR\lib\Pothos\modules0.5\proxy\environment\PythonSupport.dll): failed safe load*

This error? Do you know what options you checked when you ran the installer?

ChristoRei, 2019-04-15:
Yes, its this Error Code:

[06:39:08.996000] Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport.dll): Cannot load library: C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport.dll
[06:39:09.168000] Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport2.dll): failed safe load
[06:39:09.168000] Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport3.dll): Cannot load library: C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport3.dll
[06:39:09.231293] PothosFlow.MainWindow: Reload plugins complete
[06:39:09.543000] Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport.dll): Cannot load library: C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport.dll
[06:39:09.731000] Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport2.dll): failed safe load
[06:39:09.731000] Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport3.dll): Cannot load library: C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport3.dll

I also tried reinstallint pothos several time. I did a full installation and checked all available boxes for not missing a plugin oder anything. I also always checked boxes regarding "add to path"...

I also installed python 3.6 and then used pip to install numpy...

guruofquality, 2019-04-15:
> I also tried reinstallint pothos several time. I did a full installation and checked all available boxes for not missing a plugin oder anything. I also always checked boxes regarding "add to path"...

Make sense then, these are advanced options that are by default unchecked and only available in custom install mode.

```
cpack_add_component(python2_exclusive
    DISPLAY_NAME "Pothos Python2 blocks development plugin"
    DESCRIPTION "Python2 blocks plugin for Pothos.\nInstall only Python2 or Python3 support, but not both."
    GROUP advanced
    DEPENDS runtime)

cpack_add_component(python3_exclusive
    DISPLAY_NAME "Pothos Python3 blocks development plugin"
    DESCRIPTION "Python3 blocks plugin for Pothos.\nInstall only Python2 or Python3 support, but not both."
    GROUP advanced
    DEPENDS runtime)
```

> I also installed python 3.6 and then used pip to install numpy...

The latest installer is using python3.7.

But in any case, if you are not developing python blocks then just use "Full installation" option and do not check extra boxes.

guruofquality, 2019-04-15:
ANyway, I hope that explains it. There is an issue tracker for the windows installer if something else comes up: https://github.com/pothosware/PothosSDR/

closed 2019-04-15 by guruofquality