pothosware/PothosCore #171: Pothos Flow crashes

ChristoRei, 2019-04-25:
Hello,
I'm having problems with Pothos Flow right now.
I'm trying to analyze a car key. I am using different blocks and want to display the 2FSK signal in the Logic Analyzer. 
Whenever I start the program, it crashes with "No Response". 
Pothos Flow can only be restarted if I quit it in the Task Manager before.
After the error has occurred, Pothos sometimes closes completely without an error message. 

Could this be due to the Logic Analyzer block?
I use a Wave Trigger to pass only signals from the radio key to the Logic Analyzer. When I remove the Logic Analyzer block, the flow works. 

As hardware I use a LimeSDR.
I would be very happy about help.

The Pothos Flow is attached

[433 MHz Receive Honda.zip](https://github.com/pothosware/PothosCore/files/3116096/433.MHz.Receive.Honda.zip)

guruofquality, 2019-05-26:
Sorry I have a lot of backlog to get through. Do you have an simpler example that replicates the crash? Something without the need for a live FSK signal. I don't see the crash with a noise source in place of the limesdr. Or do you happen to have a capture of FSK for a file source block to make it happen?

ChristoRei, 2019-06-27:
Hello,
the problem I had at the time didn't occur again later. However, I finally omitted the Logic Analyzer block. 

However, I currently have a problem that matches the title perfectly, and since the post is still open, I don't want to create a new one.

I'm still recording radio keys. Meanwhile this works very well.
However, I'm currently trying to record a key that transmits on 2 different frequencies. 
I just use my topology that works for one frequency and copy it. I then create a new Soapy SDR source which I have measured on the other frequency with the other channel.
I get the following errors every time I start the program:

> Detected evaluation thread lock-up. The evaluator will not function:
> EvalEngineImpl.cpp:182 void __cdecl EvalEngineImpl::evaluate(void)
>   TopologyEval.cpp:73 void __cdecl TopologyEval::update(void)
>     TopologyEval.cpp:159 void __cdecl TopologyEval::commit(void)
> 

Sometimes the thread recovers and I get the message:

Evaluation thread has recovered. Perhaps a call is taking too long.

Does this mean that the program is too complex and my laptop can't set all values fast enough, like gain, frequency etc?

As soon as I just deactivate the second Soapy SDR Source block I don't get this message anymore.

I also tried to use only one Soapy SDR block and record two channels with it, but I get the same error.

What I want to mention: Sometimes the following error occurs. 

> Failed to commit: Framework Topology Connect Error: Pothos::Topology::commit(): Framework Topology Connect Error: AddConst0.activate(): Exception: bad allocation

Whereby there is to say here that the error does not always come from the block AddConst. The cause for this reason varies. It is almost every time a different block. 

Last but not least the last problem. 

> The block ClockRecovery sometimes throws the following error message:
> ClockRecoveryMM1[0] overconsumed 480 bytes, 460 available

Here the number of bytes varies. What is the reason? Do I have to decimate the signal first?

I would be very happy about help.

ChristoRei, 2019-07-08:
Hello,
unfortunately I have not found a solution for the problem yet.
Has anyone had similar problems?

Greetings