pothosware/PothosCore #173: Problems with GNURadio Blocks

ChristoRei, 2019-05-20:
Hello,
I have problems with some GNURadio blocks and wanted to ask if this is known and what you can do about it.

For example, the "Viterbi Combo" block is causing problems. 
After I have filled in all fields (and they are shown as green) I get an error message from the block which reads as follows:

ViterbiCombo0::eval(...) -Exception:ProxyBlockEval factory(/gr/trellis/viterbi_combined)->{Object Convert Error: Pothos::object::convert(): doesnt support int to class std::vector<class std::complex<float>,class std::allocator<class std::complex<float>>}

These error messages also occur with other GNURadio blocks like the Polyphase Synthesizer or the correlate and sync.

[image]

If someone knows these errors and could help me, I would be very grateful.

Kind Regards
ChristoRei

guruofquality, 2019-05-26:
> These error messages also occur with other GNURadio blocks like the Polyphase Synthesizer or the correlate and sync.

I would have to see the specific error, are you sure the types are correct. But Polyphase Synthesizer should be fine

For the viterbi combo, I can say that the constellation arg is wrong, it needs a list of constellation points, not a single number.

Unfortunately for the trellis blocks specifically, I did not wrap the gr::trellis::fsm class, so the part that wont work for you is FSM args.

guruofquality, 2019-05-26:
I made a new issue for the fsm issue: https://github.com/pothosware/gr-pothos/issues/13

closed 2019-05-26 by guruofquality