pothosware/PothosCore #174: MathBlocks build error on Fedora (`@97% - [Makefile:163: all] Error 2`)

mariobaldini, 2019-05-27:
Hi all, 

While trying to build and install I faced the following build error. Could someone guide me how to fix it?

```
(...)
/usr/include/c++/9/complex: In instantiation of ‘std::complex<double>& std::complex<double>::operator+=(const std::complex<_Tp>&) [with _Tp = long int]’:
/home/mario/code/PothosCore/comms/utility/SignalProbe.cpp:156:49:   required from ‘void SignalProbe<Type, ProbeType>::work() [with Type = std::complex<long int>; ProbeType = std::complex<double>]’
/home/mario/code/PothosCore/comms/utility/SignalProbe.cpp:122:10:   required from here
/usr/include/c++/9/complex:1327:13: error: no match for ‘operator+=’ (operand types are ‘std::complex<double>::_ComplexT’ {aka ‘__complex__ double’} and ‘std::complex<long int>’)
 1327 |    _M_value += __z.__rep();
make[2]: *** [comms/utility/CMakeFiles/UtilityBlocks.dir/build.make:73: comms/utility/CMakeFiles/UtilityBlocks.dir/SignalProbe.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4083: comms/utility/CMakeFiles/UtilityBlocks.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 97%] Building CXX object comms/math/CMakeFiles/MathBlocks.dir/TestScale.cpp.o
[ 97%] Building CXX object comms/math/CMakeFiles/MathBlocks.dir/Rotate.cpp.o
[ 97%] Building CXX object comms/math/CMakeFiles/MathBlocks.dir/TestRotate.cpp.o
[ 97%] Building CXX object comms/math/CMakeFiles/MathBlocks.dir/Version.cpp.o
[ 97%] Building CXX object comms/math/CMakeFiles/MathBlocks.dir/MathBlocksDocs.cpp.o
[ 97%] Linking CXX shared module libMathBlocks.so
[ 97%] Built target MathBlocks
make: *** [Makefile:163: all] Error 2
[mario@localhost build]$ 
```

**Full details:**
(following https://github.com/pothosware/PothosCore/wiki/BuildGuide)

**Environment**:
```
[mario@localhost build]$ cat /etc/fedora-release 
Fedora release 30 (Thirty)
[mario@localhost build]$ uname -a
Linux localhost.localdomain 5.0.17-300.fc30.x86_64 #1 SMP Mon May 20 15:36:26 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
[mario@localhost build]$ 

Installed libs and versions: https://pastebin.com/mzcugkKh
```

**Dependencies install:** 
```
[mario@localhost code]$ sudo dnf -y install cmake g++ numactl-devel \
>     python3 python3-numpy \
>     qt5-qtbase-devel qt5-qtsvg-devel qwt-qt5-devel \
>     portaudio-devel-19 jack-audio-connection-kit-devel graphviz
``` 

**Build cmd:** 
```
mkdir build
cd build
cmake ..
make -j4 
```

Build goes correctly up to 97%, with the above mentioned error at MathBlocks. 

Thank you in advance, 

Mario

guruofquality, 2019-05-27:
Its probably some compiler oddity. Actually it looks like your copy of the std library is missing an overload: https://en.cppreference.com/w/cpp/numeric/complex/operator_arith

The error says:

> error: no match for ‘operator+=’ (operand types are ‘std::complex<double>::_ComplexT’ {aka ‘__complex__ double’} and ‘std::complex<long int>’)

But the docs say this should be available:

```
template<class X>
complex& operator+=(const std::complex<X>& other);
```

So I'm not sure why that is happening, but the quick fix is to comment out the ifTypeDeclareFactory lines in that file for all of the types that will not compile.

guruofquality, 2019-05-31:
* see https://github.com/pothosware/PothosComms/issues/30

closed 2019-05-31 by guruofquality