pothosware/PothosCore #177: Pothos::Object conversion (std::string <-> enum class) fails in test

ncorgan, 2019-07-20:
I have a type conversion registered in "/object/convert" between a std::string and an enum class. As expected, Pothos::Object::canConvert() returns true in the static block immediately after registration.

I have a unit test for this conversion, but in the test itself, Pothos::Object::canConvert() fails. Oddly enough, Pothos::PluginRegistry::exists() confirms that the relevant converter is registered.

This is repeatable for a given enum class but inconsistent between enum classes. There's another enum class in the same test file that converts as expected.

guruofquality, 2019-07-22:
@ncorgan Its a bug in kfr, the speaker.hpp header does not have any things included that define CMT_ARCH_NAME, so the namespace is messed up in your registration file. Either add something like `#include "../kfr.h"` to speaker.hpp (which I would suggest for an upstream patch). Or add `#include <kfr/kfr.h>` before including speaker.hpp

Also you will have issues with this because MONO and M are actually aliases (the same value):
```

    STRING_ENUM_COMPARE("MONO",           Speaker, Mono);
    STRING_ENUM_COMPARE("M",              Speaker, M);
```

And just be aware since the licensing on kfr may not match your project (not 100% sure): https://opensource.stackexchange.com/questions/1777/why-is-gplv2-incompatible-with-gplv3

ncorgan, 2019-07-22:
Confirmed on my end, thanks for looking into it. These inline namespaces have been a pain in my neck.

And thanks for the other heads-ups. I've only been working with Clang since that's what they optimized for (5x compile time otherwise), and somehow I missed that it's GPLv2 and not v3.

closed 2019-07-22 by ncorgan