pothosware/PothosCore #211: Installation for Ubuntu20.04

Pradyumna007, 2020-09-06:
Am really looking to install Pothos for ubuntu-20.04. I do not see any resources. Can someone help me with this? How can I install pothos for ubuntu20.04

guruofquality, 2020-09-06:
I have not updated the PPA since I was having issues with my PGP key stuff
https://github.com/pothosware/PothosCore/issues/196

But building from source is a good option too
https://github.com/pothosware/PothosCore/wiki/BuildGuide

shredEngineer, 2021-04-20:
As building from source takes VERY long, I would also appreciate if the pothosware PPA could be made to work with Ubuntu 20.04!

plambrechtsen, 2021-05-22:
Any update on this? Or building from source is still the only option?

see613, 2021-06-26:
Any good news about the PPA for Ubuntu 20.04?

JuliDi, 2021-07-03:
Would be very helpful to have the PPA for Ubuntu 20.04. Any chance that this gets updated soon?

see613, 2021-07-03:
Maybe we need to start a petition for updating of the PPA. How many signatures would you like to see, guruofquality ?

guruofquality, 2021-07-04:
I uploaded 20.04 debs to the ppa. Any volunteers?

JuliDi, 2021-07-04:
Just installed it on 20.04 and it seems to work (installation without any errors and I can open the GUI). Thank you!