pothosware/PothosCore #217: Cannot convert Object of type long long to long

ncorgan, 2020-12-09:
"long long" declaration: https://github.com/pothosware/PothosCore/blob/master/lib/Object/Builtin/ConvertIntegers.cpp#L9
Macro: https://github.com/pothosware/PothosCore/blob/master/lib/Object/Builtin/ConvertCommonImpl.hpp#L152
"long" typedef: https://github.com/pothosware/PothosCore/blob/master/lib/Object/Builtin/ConvertCommonImpl.hpp#L26

In theory, with this, conversion should be supported between "long long" and "long", but it errors out.

guruofquality, 2020-12-12:
like this fails for you?
```
POTHOS_TEST_BLOCK("/object/tests", test_long_long_to_long)
{
    Pothos::Object obj((long long)1234);
    auto out = obj.convert<long>();
    printf("out = %d\n", (int)out);
}
```

ncorgan, 2020-12-16:
Closing issue. Taking another look at the error message, it's from Object::extract(), where the type has to be exact.

closed 2020-12-16 by ncorgan