pothosware/PothosCore #224: Build instructions for Windows

keck9939, 2021-01-15:
The build instructions on the wiki for Windows seem to be somewhat incomplete. I have been able to get Pothos to build and install using the following:
`cmake -DCMAKE_TOOLCHAIN_FILE=E:\Dev\vcpkg\scripts\buildsystems\vcpkg.cmake -DSoapySDR_DIR="E:\Program Files\PothosSDR\cmake" -Wno-dev .. -DCMAKE_INSTALL_PREFIX=E:\TEST`

where I'm using vcpkg to provide the Qt5 dependency and defining SoapySDR_DIR this way is a cheat. The -Wno-dev is needed to avoid being overwhelmed by error messages that are not intended for the end user.

`cmake  --build . --config RelWithDebInfo`

then works.

`cmake --install . --config RelWithDebInfo`

also works.

Trying to run pothosflow then fails due to missing dlls, but after copying them all from the vcpkg binary directory pothosflow starts, but puts up an error dialog with this message: "_This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem._"

I'm stuck here.

Since it may help, the output from pothosutil --info is:

PS E:\TEST\bin> .\PothosUtil.exe --system-info
Lib Version: 0.7.1-gdbef61e8
API Version: 0.7.0
ABI Version: 0.7-2
Root Path: E:\TEST\
Data Path: E:\TEST\share\Pothos
User Data: C:\Users\keck.MyDomain\AppData\Roaming\Pothos
User Config: C:\Users\keck.MyDomain\AppData\Roaming\Pothos
Runtime Library: E:\TEST\bin\Pothos.dll
Util Executable: E:\TEST\bin\PothosUtil.exe
Dev Include Path: E:\TEST\include
Dev Library Path: E:\TEST\lib
Module Search Paths:
 * E:\TEST\lib\Pothos\modules0.7-2

I am not a cmake expert, but on the other hand, it does seem like it might be possible to update the wiki to provide instructions that allow one to build a working installation without being a guru.

guruofquality, 2021-01-16:
I totally get how much of a tasking building windows and herding DLLs can be. I have put a lot of work on this branch for msvc 2019 (https://github.com/pothosware/PothosSDR/tree/msvc_2019), and it should have way less required build dependencies. So if you are currently able to build PothosCore, you should be able to build the whole environment (without the optional dependencies too). It might go easier for you: 

* https://github.com/pothosware/PothosSDR/blob/msvc_2019/README.md#required-packages-to-install
* https://github.com/pothosware/PothosSDR/blob/msvc_2019/README.md#building-the-environment

keck9939, 2021-01-16:
Thank you. I’ll give it a shot. And thank you for all the work you are doing and have done on this project.

 

*	https://github.com/pothosware/PothosSDR/blob/msvc_2019/README.md#required-packages-to-install
*	https://github.com/pothosware/PothosSDR/blob/msvc_2019/README.md#building-the-environment

eabase, 2025-02-23:
I'm currently updating the very bad build instructions here:
https://github.com/pothosware/SoapySDR/

Check out the issues...