pothosware/PothosCore #227: MSVC JIT compiler hangs if compilation fails

ncorgan, 2021-02-02:
In some circumstances (TBD), the batch script invoked by the module doesn't exit.

Ctrl+C leads to "Terminate batch job (Y/N)?", and even after entering Y, another Ctrl+C is necessary to exit.

guruofquality, 2021-06-05:
If I modify the unit test "test_compiler" with an error, I dont see the terminate issue. What conditions were you running this in?

ncorgan, 2021-06-05:
This came up in a linker error, but I'm not sure why that made a difference. Apply the attached patch and run `/util/tests/test_simd_dispatcher_utils_dump_header`. Running the generated batch script manually terminates as expected, but it hangs when run through the unit test.

[patch.txt](https://github.com/pothosware/PothosCore/files/6603107/patch.txt)

guruofquality, 2021-06-06:
So from my observation, the hang is from the output pipe getting full since there is so much verbose output from cl.exe. Eventually the compiler hangs waiting for the output pipe to be drained. A quick hack to loop over the outPipe (inside an input pipe stream) and keep reading it until its closes lets the process finish.

So I think the fix is to read in a loop into a temporary buffer instead of waiting on the process -- or to redirect cl.exe output to a file which is read out instead for the error.

guruofquality, 2021-06-07:
This branch compiler_support_read_pipe_fix should have the fix. I thought rather than merge it, we should just dogpile it on the other compiler related changes.

closed 2021-07-03 by guruofquality