pothosware/PothosCore #238: SoapySDR module directory

patvdleer, 2021-08-21:
Currently running Ubuntu 20.04.1 LTS, I installed all via the PPA's from https://github.com/pothosware/PothosCore/wiki/Ubuntu#add-ppas-to-your-system

I ran into issues with the RTL SDR, no devices found. Diving into I found the following:

```
$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.1-g33be191d
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.7 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

This while having `soapysdr0.7-module-rtlsdr` installed

The directory `/usr/local/lib/SoapySDR/modules0.7` was indeed missing, the files are located at `/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7` 

```sh
sudo mkdir -p /usr/local/lib/SoapySDR
sudo ln -s /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7 /usr/local/lib/SoapySDR/modules0.7
``` 

seems to be a temp fix

guruofquality, 2021-08-21:
If you install SoapySDR from source, it wont look for the modules from the package manager. You seem to have soapysdr library build from source, but rtl module from the package manager. https://github.com/pothosware/SoapySDR/wiki/ConfigGuide#when-to-build-from-source

patvdleer, 2021-08-22:
I got everything via apt and the ppa, I didn't compile anything myself